sld-columbia/esp #5: Add interrupt support for ESP+Ariane

davide-giri, 2019-09-18:


paulmnt, 2019-11-20:
The interrupt controller is now centralized and interrupts are delivered through the NoC. We still have direct wires between the interrupt controller (PLIC) and the CPU local interruptor (CLINT) to the processor cores. We want to have Linux SMP working before making this change.

paulmnt, 2020-09-21:
Feature implemented and merged: #58

closed 2020-09-21 by paulmnt

Kendidi, 2020-10-02:
@paulmnt and @davide-giri 

I am encountering an issue where when booting ESP bbl+Kernel with our Ariane bitfile (single core, built according to CVA6), it always stuck at instruction "wfi" as shown below in head.S.

    .align 2
    .Lsecondary_park:
    	/* We lack SMP support or have too many harts, so park this hart */
    	wfi
    	j .Lsecondary_park

I do not see this issue with the ESP environment.  

I think this may be be related to the SMP issue you have seen before.  Would you please shed some light on how I should tackle this issue?  Thanks a lot in advance!

paulmnt, 2020-10-02:
SMP is not working on ESP yet, because Ariane's L1 is cannot be invalidated from AXI. We need to implement that interface in Ariane first.

Having said that, there is no reason why Linux SMP should not work on a single core built. That might be related to a bug in your current build.

Kendidi, 2020-10-02:
I see.  Thank you @paulmnt !