sld-columbia/esp #6: Accelerator design automation

paulmnt, 2019-10-03:
- Automatically generate the accelerator device driver from existing templates.
- Automatically generate XML description of the accelerator.
- Generate the skeleton for user-space application based on libesp.
- Generate working skeleton for load and store functions.

seldridge, 2019-10-03:
This should probably be an issue for the Chisel accelerator, too. It can do 1/4 of these currently, but there's no reason why it can't do them all, too.

paulmnt, 2019-10-03:
Right now I have a script for SystemC and C based accelerators and this should be able to generate Chisel code as well in the future.
Were you suggesting to have a separate script in Chisel that generates the driver?
It's possible, but since the device driver won't change I don't see the need for duplicating the code.
Would you be ok if we extend this script to generate the skeleton for a Chisel accelerator as well?
This can be done once we have a more general example of Chisel accelerator with DMA capabilities.

seldridge, 2019-10-03:
I think that's fine initially. However, I would want a full solution to be type safe in the same way that the XML generation from Chisel is.

paulmnt, 2019-10-03:
I see. We should spec this out starting from the full example.

I will push the script soon. With respect to application and device driver there is very little to change from the template and I would rather have this common to all design flows (Chisel, SystemC, C).

The generation of load and store functionality, instead, can be different for Chisel accelerators. The whole point is to make it simpler for users to create the accelerator. Ideally, users should not be required to edit load and store at all. At the moment this is true for simple data structures only, but I believe it can be generalized.

paulmnt, 2019-10-03:
First draft of the script is available on the branch 'devel'

seldridge, 2019-10-03:
Maybe throw up a PR or a draft PR. That might make it easier to discuss/solicit feedback.

paulmnt, 2019-10-14:
SystemC flow and Vivado HLS flows are merged.
The Vivado HLS flow still requires fixes for some corner case at the interface.

jzuckerman, 2022-03-08:
New Catapult flows are TODO here.