sld-columbia/esp #15: NVDLA DMA performance

paulmnt, 2019-11-14:
NVDLA is not leveraging AXI burst: all transactions have length set to 1 beat.
This is not ideal to communicate over the NoC with the current implementation of the AXI proxy.

We need to allow the AXI proxy to handle multiple outstanding transactions. The number of transactions should be configurable at design time and result in the instance of a queue to hold the necessary information to process DMA read responses.

In addition, the DMA engine must be split into two separate processes to handle requests and responses in parallel.

davide-giri, 2020-05-04:
To mitigate the issue we increased the NVDLA burst length from 1 beat to 4, that's the maximum allowed in the NVDLA configuration files. This feature is currently in the `nvdla` branch only, but it will be merged into `master` soon.

paulmnt, 2020-09-21:
Merged with #46.
We will still work on improving DMA over AXI with other examples of third-party accelerators.

closed 2020-09-21 by paulmnt