sld-columbia/esp #19: Device drivers incorrectly configure accelerators with >4 config parameters

davide-giri, 2019-11-25:
**Describe the bug**
Device drivers incorrectly configure accelerators with more that 4 configuration parameters. The p2p-related functions in the driver most likely have a buffer overflow.

**To Reproduce**
Create any ESP accelerator with more than four configuration parameters and test it on FPGA.

**Expected behavior**
Currently ESP works with up to 14 custom configuration parameters per accelerator.

paulmnt, 2020-02-10:
I can't reproduce the issue described, but there was a bug in the Linux application for the synthetic accelerator that could cause the same behavior described in this issue: if your application does not set any value for the point-to-point fields of the ESP descriptor, the corresponding registers could be erroneously initialized.

Can you please check whether this fixes your issue?

This has been resolved for the synthetic accelerator:

```patch
commit de914c40951629a1c8751bb4452fa2ed3dbc4acc
Author: Paolo Mantovani <paolo@cs.columbia.edu>
Date:   Wed Jan 29 12:26:01 2020 -0500

    :bug: disable p2p for soft/leon3/drivers/synth/app/synth.c

diff --git a/soft/leon3/drivers/synth/app/synth.c b/soft/leon3/drivers/synth/app/synth.c
index 40a932a..02b88fb 100644
--- a/soft/leon3/drivers/synth/app/synth.c
+++ b/soft/leon3/drivers/synth/app/synth.c
@@ -355,6 +355,8 @@ static void alloc_phase(accelerator_thread_info_t **info, int nthreads,
 
                        info[i]->chain[acc].desc.esp.alloc_policy = params.policy;
                        info[i]->chain[acc].desc.esp.ddr_node = contig_to_most_allocated(info[i]->mem);
+                       info[i]->chain[acc].desc.esp.p2p_store = 0;
+                       info[i]->chain[acc].desc.esp.p2p_nsrcs = 0;
                }
        }
 }
```

paulmnt, 2020-04-02:
Closing the issue as not observable. Please reopen if necessary.

closed 2020-04-02 by paulmnt