sld-columbia/esp #23: Incorrect sporadic behavior of Vivado HLS accelerators

davide-giri, 2020-01-13:
**Describe the bug**
When testing ESP on FPGA, for an unknown reason sometimes the Vivado HLS accelerators do not function correctly. When adding debug probes with the Vivado ILA utility to the accelerator interface, the accelerators always function correctly.

davide-giri, 2020-01-13:
**Problem**
It appears Vivado optimized away some signals at the interface of the Vivado HLS accelerator.

**Solution**
We solved the bug by adding the attribute `keep` to the signals at the interface of the Vivado HLS accelerator.

davide-giri, 2020-04-03:
Reopening issue: Although the previous fix has reduced the bug occurrences, we can still sporadically observe the bug.

paulmnt, 2020-05-13:
This bug has been fixed by commits 15ded13 and 07aa706.

  - DMA control channels are now implemented using Vivado HLS `ap_hs` interface.
  - `ap_start` depends on `ap_ready`, in compliance with the Vivado HLS `ap_ctrl_hs` interface.
  - DMA `valid` and `ready` control signals are now independent.
  - FIFOs between Vivado HLS accelerator and ESP interface have been removed.

closed 2020-05-13 by paulmnt