sld-columbia/esp #26: ESP xconfig NoC dimension constraint checking

zzhu35, 2020-02-13:
The ESP xconfigurator says that the maximum supported number of tiles is 64. However the number of bits used to index the tiles in nockpackage is 5 bits. https://github.com/sld-columbia/esp/blob/master/rtl/include/sld/noc/nocpackage.vhd#L36
If someone instantiates an SOC of 64 X 1, ESP xconfigurator lets it pass but the NOC might get confused.

Bug already reported in email to
Paolo Mantovani <paolo@cs.columbia.edu> 
Davide Giri <davide.giri@columbia.edu>

paulmnt, 2020-02-13:
Thank you for reporting this issue. We should improve the router's design to have a parameter define the number of bits needed for coordinates on the network, as well as add a check in the configuration GUI for ESP.
In the meantime, please select up to 8 tiles per dimension (8x8 is the largest configuration currenlty supported), even if the GUI lets you choose more.

Once we close this issue, you will be able to scale up to 32x32 tiles, by using all of the 5 bits reserved for each coordinate.

At the moment, we are not planning to increase the packet header size from one to two flits to support a larger flat 2-D mesh. Rather, we are considering a hierarchical design to further improve scalability of large instances of ESP.

davide-giri, 2020-03-11:
We now allow at most 8 rows and 8 columns in the GUI, so the bug has been fixed (see commit 
c71f5db)

I'm changing the label from bug to enhancement. The maximum number of rows and columns is still hard-coded in the NoC routers, it should become configurable, allowing up to 32 rows and 32 columns.