sld-columbia/esp #27: Synthesis Pipeline Broken

zzhu35, 2020-02-21:
After pulling the most recent commit https://github.com/sld-columbia/esp/commit/794a0a99b9601a31a25e5bcb472d209b225bf497 from master to a local clean install of ESP, Vivado synthesis pipeline errors out with 
```
ERROR: [Synth 8-1031] cache_attribute_array is not declared [/projects/esp/socs/xilinx-vc707-xc7vx485t/socmap.vhd:290]
ERROR: [Synth 8-1031] dma_attribute_array is not declared [/projects/esp/socs/xilinx-vc707-xc7vx485t/socmap.vhd:349]
ERROR: [Synth 8-1031] cache_tile_id is not declared [/projects/esp/socs/xilinx-vc707-xc7vx485t/socmap.vhd:463]
ERROR: [Synth 8-1031] cache_tile_id is not declared [/projects/esp/socs/xilinx-vc707-xc7vx485t/socmap.vhd:464]
ERROR: [Synth 8-1031] cache_tile_id is not declared [/projects/esp/socs/xilinx-vc707-xc7vx485t/socmap.vhd:465]
ERROR: [Synth 8-1031] cache_tile_id is not declared [/projects/esp/socs/xilinx-vc707-xc7vx485t/socmap.vhd:466]
ERROR: [Synth 8-1031] cache_tile_id is not declared [/projects/esp/socs/xilinx-vc707-xc7vx485t/socmap.vhd:467]
ERROR: [Synth 8-1031] cache_tile_id is not declared [/projects/esp/socs/xilinx-vc707-xc7vx485t/socmap.vhd:468]
ERROR: [Synth 8-1031] cache_tile_id is not declared [/projects/esp/socs/xilinx-vc707-xc7vx485t/socmap.vhd:469]
ERROR: [Synth 8-1031] cache_tile_id is not declared [/projects/esp/socs/xilinx-vc707-xc7vx485t/socmap.vhd:470]
ERROR: [Synth 8-1031] cache_tile_id is not declared [/projects/esp/socs/xilinx-vc707-xc7vx485t/socmap.vhd:471]
ERROR: [Synth 8-1031] cache_tile_id is not declared [/projects/esp/socs/xilinx-vc707-xc7vx485t/socmap.vhd:472]
ERROR: [Synth 8-1031] cache_tile_id is not declared [/projects/esp/socs/xilinx-vc707-xc7vx485t/socmap.vhd:473]
ERROR: [Synth 8-1031] cache_tile_id is not declared [/projects/esp/socs/xilinx-vc707-xc7vx485t/socmap.vhd:474]
ERROR: [Synth 8-1031] cache_tile_id is not declared [/projects/esp/socs/xilinx-vc707-xc7vx485t/socmap.vhd:475]
ERROR: [Synth 8-1031] cache_tile_id is not declared [/projects/esp/socs/xilinx-vc707-xc7vx485t/socmap.vhd:476]
ERROR: [Synth 8-1031] cache_tile_id is not declared [/projects/esp/socs/xilinx-vc707-xc7vx485t/socmap.vhd:477]
ERROR: [Synth 8-1031] cache_tile_id is not declared [/projects/esp/socs/xilinx-vc707-xc7vx485t/socmap.vhd:478]
ERROR: [Synth 8-1031] cache_tile_id is not declared [/projects/esp/socs/xilinx-vc707-xc7vx485t/socmap.vhd:480]
```
After reverting to an older commit https://github.com/sld-columbia/esp/commit/c462c99bd005e07a1d94bb4f485980d8c463576f, everything is working fine. I did not try any version in between though.

davide-giri, 2020-02-22:
Hi, have you run `make distclean` after the `git pull` in the design folder? That should do it.

We added a feature to the ESP GUI from which `socmap.vhd` is generated. You probably had a local version of `socmap.vhd` from before this update.

zzhu35, 2020-02-23:
Yes, `make distclean` fixed it. Thanks!

closed 2020-02-23 by zzhu35