sld-columbia/esp #29: Need help simulating ESP

notooth1, 2020-03-12:
Hello,

I got this error when simulating ESP. Can anyone help?
> ** Error: /mnt/Archive/Downloads/esp/rtl/src/techmap/unisim/tap_unisim.vhd(30): (vcom-1598) Library "unisim" not found.
** Error: /mnt/Archive/Downloads/esp/rtl/src/techmap/unisim/tap_unisim.vhd(31): (vcom-1136) Unknown identifier "unisim".
** Note: /mnt/Archive/Downloads/esp/rtl/src/techmap/unisim/tap_unisim.vhd(34): VHDL Compiler exiting
make: *** [/mnt/Archive/Downloads/esp/utils/Makefile:1174: modelsim/vsim.mk] Error 2

davide-giri, 2020-03-12:
Hi,

It seems that something got broken in your design folder. If you are fine deleting your local changes I would run `make distclean` in the subfolder of `socs` that you are working in, then make sure to setup all the environment variables and finally repeat all the steps cleanly.

If that doesn't work, can you mention which FPGA you are targeting and could you attach a screenshot of the ESP GUI, you can open it with `make esp-xconfig`.

Thanks

notooth1, 2020-03-12:
`make distclean` did not solve the problem. The FPGA target is xilinx-vc707-xc7vx485t. Here is the screenshot of ESP GUI: [xilinx-vc707-xc7vx485t]screenshot

davide-giri, 2020-03-12:
It seems you are using the default configuration for that FPGA, so no issues there.

The problem you are having is normally caused by some Xilinx libraries not being compiled correctly. The command you launched is `make sim` or `make sim-gui`, right? If you look at what happened right after launching it you should see some errors regarding Xilinx libraries. The cause may be:
- Environment variables not setup correctly. Do you have Vivado in PATH?
- Using the wrong version of Vivado. ESP requires Vivado 2018.2.

You may want to `make distclean` again before trying this out, just to be safe.

I hope this helps

notooth1, 2020-03-13:
Yes, the simulating command is `make sim`
There is no error after launching Vivado. Here is my environment variables:

> export VIVADO=/home/notooth/Xilinx/Vivado/2019.2
. $VIVADO/settings64.sh
export leon=/home/notooth/Downloads/esp-leon
export PATH=$PATH:$leon/bin
export PATH=$PATH:$leon/mklinuximg
export PATH=$PATH:$leon/sparc-elf/bin

davide-giri, 2020-03-13:
As I was saying in the previous comment, ESP requires Vivado 2018.2. I see that you are using Vivado 2019.2, so that should be the problem.

notooth1, 2020-03-13:
I have just installed Vivado 2018.2, and set the environment variables like this:

> export VIVADO=/home/notooth/Xilinx/Vivado/2018.2
. $VIVADO/settings64.sh
export leon=/home/notooth/Downloads/esp-leon
export PATH=$PATH:$leon/bin
export PATH=$PATH:$leon/mklinuximg
export PATH=$PATH:$leon/sparc-elf/bin

After running the command `make distclean` and `make sim`, I still get the same error.

davide-giri, 2020-03-13:
`. $VIVADO/settings64.sh`: this may be the problem. You need to source it like this: `source $VIVADO/settings64.sh`. It may also be some other environment problem. The decription of how to set the environment variables is in this guide: https://www.esp.cs.columbia.edu/docs/setup/setup-guide/

In the meantime could you share the full terminal printouts when you run `make sim`?

peter-d, 2020-04-16:
I had the same problem. Seems things go wrong on the build of the unisim libs:
```vivado -mode batch -notrace -source xilinx_lib/simlib.tcl``` 
(not sure where in the build flow this runs exactly though)

Are you using questasim instead of modelsim? 

[This Xilinx forum post](https://forums.xilinx.com/t5/Simulation-and-Verification/Error-when-compiling-xilinx-libraries-for-modelsim/m-p/685942/highlight/true#M15484) suggests to change the `compile_simlib` command arguments. That gets me one step further, but still givers a Vivado error.

paulmnt, 2020-04-16:
This error is not related to ESP specific files, but since we use Xilinx BRAMs to implement memories, the `make sim` target invokes Vivado to compile the simulation libraries as a first step.

This step will fail if the versions of Vivado and Modelsim are incompatible. I believe that Vivado 2018.2 recommends Modelsim 10.6, but we are simulating successfully with Modelsim 10.7a as well. Older versions of Modelsim are likely to fail.

As @peter-d suggests, if you are using Questasim, you should still be able to simulate, as long as the version of Questa you have is compatible with Vivado 2018.2.

To use Questa, edit the following line:
https://github.com/sld-columbia/esp/blob/master/utils/Makefile#L888

```patch
diff --git a/utils/Makefile b/utils/Makefile
index 7955064..b91e234 100644
--- a/utils/Makefile
+++ b/utils/Makefile
@@ -885,7 +885,7 @@ check_all_rtl_srcs-distclean:
 
 modelsim/xilinx_lib: $(XILINX_VIVADO)
        $(QUIET_MKDIR)mkdir -p $@
-       @echo "compile_simlib -directory xilinx_lib -simulator modelsim -library unisim -no_ip_compile" > $@/simlib.tcl; \
+       @echo "compile_simlib -directory xilinx_lib -simulator questa -library unisim -no_ip_compile" > $@/simlib.tcl; \
        cd modelsim; \
        vivado $(VIVADO_BATCH_OPT) -source xilinx_lib/simlib.tcl; \
        lib_path=$$(cat modelsim.ini | grep secureip | cut -d " " -f 3); \
```
After patching the Makefile, from the design folder `socs/<target_fpga_board>` clean with `make sim-distclean` before using `make sim` again.

@peter-d, which is the onther error you get after changing the target simulator to modelsim?

If you don't have access to a compatible version of Modelsim, or Questa, you may try Incisive (Leon3 only), or Xcelium from Cadence as well.
If you use Xcelium and simulate using Ariane, please follow the instructions in the section **Patching Ariane for Xcelium Simulator** here:
https://esp.cs.columbia.edu/docs/setup/setup-guide/

peter-d, 2020-04-16:
Thank you @paulmnt 
The next error I get is [this one](https://forums.xilinx.com/t5/Simulation-and-Verification/Vivado-2018-3-compiled-simulation-libraries-have-novopt-option/td-p/961721)

I have vivado 2017.2 and not an old enough questasim (10.7c)
I will need to upgrade/downgrade to a compatible set of tools.

I have ncsim as well, but the default environment set up in my org for that includes a gcc provided with the tool that gets in my path, which messes up the firmware build part of the scripts - so I prefer to get things working with questasim, as that would support both leon3 and ariane cores.

paulmnt, 2020-04-16:
I see: the problem for you is the `-novopt` option.

[This Xilinx AR](https://www.xilinx.com/support/answers/68324.html) lists which version of Modelsim is supported officially by each version of Vivado. If you can get any of these pairs, you should be able to simulate.

I believe you can download any version of Vivado for free and run simulation. You only need a license for synthesis and implementation.

If you need to run synthesis on Vivado 2017.2, you can checkout mig.prj, mig.xci and sgmii.xci  from the constraint folder `<esp>/constraints/<target_fpga>` at the commit `1a884047890635ffe10af624bcbd6c38d2027dc7`.
Checkout **only** the Xilinx IP files.

At the time we had Vivado 2017.2 and Modelsim 10.5c. Please note that Ultrascale and Ultrascale+ boards were not supported. Also, I cannot guarantee that things will work out of the box if you revert the IP files to that commit, but even if you get errors, you should be able to get past them with minor fixes.

At some point, we may also bump the version of Vivado to 2019.2 on ESP master, but we typically do that only when there is a major issue that a new version of Vivado resolves (e.g. bug fixes, performance improvements, etc.).

Please let us know if any of the above works for you.

peter-d, 2020-04-21:
Hi Paul, I’ve managed to get it all up and running with Vivado 2018.2 and the corresponding questasim version. Thank you for the help!

paulmnt, 2020-04-21:
Great to hear that! thank you for letting us know!

@notooth1 were you able to run the RTL simulation as well?

closed 2020-06-04 by davide-giri

z0CoolCS, 2022-08-12:
@peter-d could you share your solution please? I am facing the same problem with -novopt.

peter-d, 2022-08-12:
> @peter-d could you share your solution please? I am facing the same problem with -novopt.

It’s a matter of making sure Vivado and modelsim/questa versions match with what Xilinx prescribed,  like @paulmnt said earlier in the thread (see the Xilinx [AR](https://www.xilinx.com/support/answers/68324.html)  he linked)

MutaharKhadim0, 2024-07-28:
> This error is not related to ESP specific files, but since we use Xilinx BRAMs to implement memories, the `make sim` target invokes Vivado to compile the simulation libraries as a first step.
> 
> This step will fail if the versions of Vivado and Modelsim are incompatible. I believe that Vivado 2018.2 recommends Modelsim 10.6, but we are simulating successfully with Modelsim 10.7a as well. Older versions of Modelsim are likely to fail.
> 
> As @peter-d suggests, if you are using Questasim, you should still be able to simulate, as long as the version of Questa you have is compatible with Vivado 2018.2.
> 
> To use Questa, edit the following line: https://github.com/sld-columbia/esp/blob/master/utils/Makefile#L888
> 
> ```diff
> diff --git a/utils/Makefile b/utils/Makefile
> index 7955064..b91e234 100644
> --- a/utils/Makefile
> +++ b/utils/Makefile
> @@ -885,7 +885,7 @@ check_all_rtl_srcs-distclean:
>  
>  modelsim/xilinx_lib: $(XILINX_VIVADO)
>         $(QUIET_MKDIR)mkdir -p $@
> -       @echo "compile_simlib -directory xilinx_lib -simulator modelsim -library unisim -no_ip_compile" > $@/simlib.tcl; \
> +       @echo "compile_simlib -directory xilinx_lib -simulator questa -library unisim -no_ip_compile" > $@/simlib.tcl; \
>         cd modelsim; \
>         vivado $(VIVADO_BATCH_OPT) -source xilinx_lib/simlib.tcl; \
>         lib_path=$$(cat modelsim.ini | grep secureip | cut -d " " -f 3); \
> ```
> 
> After patching the Makefile, from the design folder `socs/<target_fpga_board>` clean with `make sim-distclean` before using `make sim` again.
> 
> @peter-d, which is the onther error you get after changing the target simulator to modelsim?
> 
> If you don't have access to a compatible version of Modelsim, or Questa, you may try Incisive (Leon3 only), or Xcelium from Cadence as well. If you use Xcelium and simulate using Ariane, please follow the instructions in the section **Patching Ariane for Xcelium Simulator** here: https://esp.cs.columbia.edu/docs/setup/setup-guide/

i am facing same issue as mentioned above and i am using questasim which make file i have to modify to use questasim. I didn't find the above given lines in Makefile in utils directory