sld-columbia/esp #31: L1 INVAL on SMAD(W)

zzhu35, 2020-04-06:
**Describe the bug**
During SMAD(W) transition, when receiving an FWD_INV from NoC, ESP L2 does not send an INVAL to in-tile L1. This can cause erroneous behavior when multicores write to different words in same cache line (MESI false sharing).

**To Reproduce**
Steps to reproduce the behavior:
1. change **utils/grlib/software/leon3/leon3_test.c** to the content in this gist https://gist.github.com/zzhu35/2c5f1a012827f3bfaad806f6ad4492cf
2. change **socs/\<design-folder\>/systest.c** to the content in the above gist as well
3. configure ESP as the picture below
4. make vivado-syn make fpga-program make fpga-run

**Expected behavior**
Core 0 should print results from the all the tests. However, system hangs after printing "Start testing on 4 CPUs." Although I am not 100% sure if this is actually caused by the reported bug, but this behavior was identical to Spandex before the above bug was fixed by inserting **inval(fwd_in.addr)** to L1.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [CentOS]
 - CAD tools versions: [Xilinx Vivado 2018.2]

**Additional context**
1) We start with a cache line shared by A and B.
2) L1A and L1B simultaneously send a write request to L2A and L2B for different words within the same cache line.
3) L2A and L2B send a GetM to the LLC and change the state S --> SMAD.
4) The LLC processes first the GetM from A. As a result: S --> M, Data to A, Invalidate to B.
5) Immediately after the LLC processes the GetM from B. As a result: FwdGetM to A.
6) L2A receives the Data. As a result: SMAD --> SMA.
7) L2B receives the Invalidate. As a result: SMAD --> IMAD and InvAck to A. Here is where we should probably send an Invalidate to L1B.
8) L2A receives the FwdGetM, but that request is stalled while waiting for the InvAck.
9) L2A receives the InvAck. As a result: SMA --> M.
10) L2A processes the stalled FwdGetM. As a result: M --> I, Invalidate to L1A, Data to B. This means that L2A and L1A are correctly invalidated.
11) L2B receives Data from A. As a result: IMAD --> M. Now L2B is updated correctly, but L1B is not.

See email thread with subject **ESP L1 Invalidations**

zzhu35, 2020-04-06:
Sorry the screenshot is cropped. All 4 cores are Leon3.

paulmnt, 2020-04-06:
Thank you for reporting this issue. Please note that if confirmed, this might affect recalls as well.
Potential affected scenarios (multithreaded applications, coherent DMA of accelerators and peripherals such as Ethernet).

davide-giri, 2020-04-07:
We can observe the bug and we were able to fix it. We will do more testing before pushing the fix and closing the issue.

zzhu35, 2020-04-07:
That is great news! Thank you for your work! I have this issue patched in Spandex but I will merge-in your patch if it's better than what I have

davide-giri, 2020-04-09:
Bug fixed with commit eb8dbad for both the SystemVerilog and the SystemC implementation of the caches. The fix was to add the missing L1 invalidation as reported in the issue description. 

This bug fix solved an existing problem with using the ethernet in multi-core ESP SoCs. Now, when using the SystemVerilog caches, operations like SSH and SCP work also for multi-core ESP SoCs.

@zzhu35 thanks for finding the bug!

closed 2020-04-09 by davide-giri