sld-columbia/esp #32: Request for documentation: building accelerators from Verilog

charmoniumQ, 2020-04-07:
**Is your feature request related to a problem? Please describe.**
I have an existing accelerator implementation in Verilog, and I would like to port it to ESP infrastructure, so that I can use the ESP toolchain with Vivado tools.

The [website][1] says
> Other options include designing the accelerator in RTL (Verilog, VHDL, SystemVerilog) or designing a deep learning accelerator in Keras/Pytorch/ONNX by leveraging hls4ml. These other options don’t have full support in ESP yet. It’s possible to use them, but they will require more integration effort.

**Describe the solution you'd like**
If there are any examples (it sounds like there might be from [this page][1]?) or advice on doing that, I would appreciate those resources. I would document and contribute back my process once I get it to work.

**Describe alternatives you've considered**
I am currently trying to work through `accelerators/vivado_hls/common/syn/common.tcl` and convert that to a version which accepts Verilog input, but this is not my strongsuit.

This is a request for documentation, not a feature request _per se_, but I am posting it here instead of in an email so that those who encounter this problem in the future can see this issue.

[1]: https://www.esp.cs.columbia.edu/teaching/

paulmnt, 2020-04-07:
Thank you for writing. We have not released a document about the integration of a third party IP from RTL, however, the NVDLA is a good example to look at.

There are two approaches to the integration of an accelerator from RTL:

1. Third-party flow (e.g. NVDLA).

Requirements:

 - accelerator has AXI master interface to access memory
- accelerator has APB or AXI-lite slave interface to let the CPU configure registers
- the accelerator designer provides driver software that can compile for either RISC-V or SPARC

Example of integration:

- NVDLA

Advantages:

- use accelerator software and RTL as is

2. Use ESP native interface

Requirements:

- accelerator must expose FIFO-based DMA interface. (We can provide you with pointers to a detailed description)
Example of integration:
- TODO

 Advantages:

- leverage ESP software stack and SoC services (e.g. scatter-gather DMA, accelerator virtual address space, ...)
- generate driver and software skeleton.

We will add a tutorial about the third-party IP flow and we may implement a simple RTL accelerator with native ESP interfaces in the future, but if you let us know which flow you would prefer to follow, we can provide further assistance on how to proceed.

Also, I think that using Vivado HLS to create an adapter for your accelerator may not be that easy. If you are not familiar with HLS, I would switch to one of the approaches above.

charmoniumQ, 2020-04-08:
I'm interested in using ESP native interface. Can you point me to the detailed description?

davide-giri, 2020-04-09:
Hi,

**Accelerator interface**

Currently our documentation doesn't provide a detailed description of the ESP accelerator interface. We will provide the specification of the interface some time next week.

In the meantime I'll share a couple of slides that describe the interface, so you can get an idea. The code in the slides refers to the SystemC implementation of the interface, but it should still give you a sense of what the interface looks like.

[ESP_accelerator_model_extract.pdf](https://github.com/sld-columbia/esp/files/4453985/Di_Guglielmo_ESP_accelerator_model_extract.pdf)

After a reset the accelerator starts checking the signal `conf_done`, when that signal becomes `1` the accelerator can start. When the accelerators is done with its task it will set the `kernel_done` signal to `1`. The configuration parameters of the accelerators are contained in the struct `conf_info`. Those configuration signals are set before conf_done and they will be constant until the accelerator sets `kernel_done`.

The accelerator sends out a read or a write request through `dma_read_ctrl` and `dma_write_ctrl` respectively. The request is made of base offset, length and word size. For read requests, the requested data arrives on the `dma_read_chnl`, while for write requests the accelerator sends out data on the `dma_write_chnl`.

`dma_read_ctrl`, `dma_write_ctrl`, `dma_read_chnl` and `dma_write_chnl` are latency insensitive channels, which means that each of them is composed by three signals: `data`, `valid` and `ready`. `data` gets transferred only when both `valid` and `ready` are `1`.

With this information you should be able to implement the ESP interface for your Verilog accelerator.

**Design flow**

The best way to integrate your accelerator in ESP would be to first generate either a Stratus HLS or a Vivado HLS accelerator by following the ESP tutorials (no need to modify the generated accelerator to become a MAC, you can keep the default generated by ESP).  This way ESP will also generate for you the device driver and a skeleton for the bare-metal and Linux test applications. 

Then you can launch the HLS so that the Verilog implementation of the accelerator will be generated in the `tech` folder. There you can replace the generated accelerator with your own. ESP will see your accelerator and you will be able to instantiate it in the ESP GUI.

I hope this helps, let us know if you have any other questions. We'll get back to you with more information next week.

Davide

paulmnt, 2020-05-04:
[image]

@charmoniumQ: we are writing up a document that describes the ESP accelerator interface in detail. In addition to the explanation of @davide-giri, I think these waveforms should help you understand how DMA transactions work. We will send the link to the complete document as soon as it's available on the ESP website.

jzuckerman, 2022-03-08:
A document exists on the accelerator interface specs on the ESP website: https://esp.cs.columbia.edu/docs/specs/esp_accelerator_specification.pdf

There's also a new flow for designing RTL accelerators included in #123

closed 2022-03-08 by jzuckerman