sld-columbia/esp #34: Leon3 Cache Flush Support

zzhu35, 2020-04-20:
Leon3 supports whole-cache-flush instructions according to section 86.3.11 in GrLib IP manual https://www.gaisler.com/products/grlib/grip.pdf How does ESP handle this "flush" CPU request?

cpu_msg is obtained from concatenating two fields, **hwrite** and **hmastlock**, of the **ahb_slv_in_type** data structure, which maps to four kinds of requests (atomic R/atomic W/R/W). I think this implies that ESP L2 (and LLC of course) is never aware of the CPU's cache-flush instructions. 

This mysterious comment line here https://github.com/sld-columbia/esp/blob/master/systemc/common/caches/cache_types.hpp#L64 indicates that you at least attempted to handle that request but then decided not to support it. Is that true? Why or why not? How exactly is the flush request passed down to the rest of the cache-hierarchy from L1?

davide-giri, 2020-04-20:
In ESP the flush of the L1 data cache gets propagated to the respective L2 private cache. You can see that in the `l2_wrapper.vhd` file that you linked in your issue description. The `l2_wrapper` module has an input called `flush` and if you look at the code you can see the details of how that gets propagated to the L2 cache. For what concerns the private caches in the accelerator tiles, you should know that they get flushed at the end of the accelerator execution.

The comment that you found in `cache_types.vhd` is a carry over from an older implementation, now there's a specific input port of the L2 called `l2_flush`.

The L1 data flush does not need to be propagated to the LLC. However, beside flushing the private caches, ESP also allows the processors to flush the LLC. The LLC flush is needed by non-coherent devices. For example an accelerator that performs non-coherent DMA requires the LLC flush before starting to execute.

In ESP, a processor core can issue an LLC flush with a write to a memory mapped register. The command arrives to the LLC wrapper on an APB bus and it gets propagated to the LLC.
https://github.com/sld-columbia/esp/blob/87524b664c4d2889075d961b36ed12cc58fa0ab5/rtl/src/sld/caches/llc_wrapper.vhd#L1681-L1684 

Here is the set of software functions that ESP makes available for flushing. These are mainly used when it comes to invoking accelerators, that is they are used by the bare-metal applications to test accelerators and by the accelerator's Linux device drivers:
* Bare-metal flush implementation: https://github.com/sld-columbia/esp/blob/87524b664c4d2889075d961b36ed12cc58fa0ab5/soft/leon3/drivers/probe/probe.c#L78
* Flush implementation for Linux
    * https://github.com/sld-columbia/esp/blob/87524b664c4d2889075d961b36ed12cc58fa0ab5/soft/leon3/drivers/esp/esp.c#L86
    * https://github.com/sld-columbia/esp/tree/master/soft/leon3/drivers/esp_cache

This is an overview of _flushing_ in ESP, but of course there's more to it. I hope this helps and let us know if you have any more questions.

Thank you

zzhu35, 2020-04-21:
Thank you for your answer!

closed 2020-04-21 by zzhu35