sld-columbia/esp #35: Accelerator L2 wrapper issues flush before all outstanding requests are served

jzuckerman, 2020-04-21:
**Describe the bug**
The accelerator L2 wrapper issues a flush to the L2 before the last requests are served. Because of the priority encoding in the cache, the flush is processed before CPU requests. This breaks coherence assumptions that an accelerator is exclusively working on data, and triggers undefined behavior with later DMA requests. 

**To Reproduce**
Steps to reproduce the behavior:
On branch esp-synth-app, create an SoC with 12 synthetic accelerators. Run the application synth.exe with the file synth_cfg.txt . 

**Expected behavior**
The flush should not be issued until the final request has been served.

jzuckerman, 2020-05-06:
Fixed on branch _synth-app_ in commit e04582 . Will update and close issue once merged into master

paulmnt, 2020-09-21:
Feature merged: #58

closed 2020-09-21 by paulmnt