sld-columbia/esp #37: Support Red Hat Linux (init_accelerator.sh)

GiuseppeDiGuglielmo, 2020-05-04:
**Describe the bug**

The script `init_accelerator.sh` is not compatible with Red Hat Linux

See for example:
https://github.com/sld-columbia/esp/blob/master/utils/scripts/init_accelerator.sh#L305-L311

We should add a test to catch `rhel`, for example:
```
    if cat /etc/os-release | grep -q -i rhel; then
        rename accelerator $LOWER *
    fi
```
and add a default that terminates the script when an unknown OS is detected.

GiuseppeDiGuglielmo, 2020-05-05:
A working but bare minimum support of Red Hat for the above script:
https://github.com/sld-columbia/esp/blob/catapult-hls/utils/scripts/init_accelerator.sh

paulmnt, 2020-09-21:
Patch applied with commit 1283c56d

closed 2020-09-21 by paulmnt