sld-columbia/esp #38: The script init_accelerator.sh can generate incorrect accelerators for Vivado HLS

lucapiccolboni, 2020-05-04:
The script `esp/utils/scripts/init_accelerator.sh` can generate accelerators that are not functionally correct when simulated with Vivado HLS. You can reproduce the bug as follows:

```
 * Enter accelerator name [dummy]: example
 * Select design flow (Stratus HLS, Vivado HLS, hls4ml) [S]: V
 * Enter ESP path [/home/piccolboni/esp]:
 * Enter unique accelerator id as three hex digits [04A]:
 * Enter accelerator registers
  - register 0 name []: in_bytes
  - register 0 default value [1]: 16
  - register 0 max value [16]: 65536
 * Configure PLM size and create skeleton for load and store:
  - Enter data bit-width (8, 16, 32, 64) [32]: 8
  - Enter input data size in terms of configuration registers (e.g. 2 * in_bytes}) [in_bytes]: in_bytes
   data_in_size_max = 65536
  - Enter output data size in terms of configuration registers (e.g. 2 * in_bytes) [in_bytes]: in_bytes
   data_out_size_max = 65536
  - Enter an integer chunking factor (use 1 if you want PLM size equal to data size) [1]: 1024
   Input PLM has 64 8-bits words
   Output PLM has 64 8-bits words
  - Enter number of input data to be processed in batch (can be function of configuration registers) [1]: 1
   batching_factor_max = 1
  - Is output stored in place? [N]: N
```

If you try to simulate it, Vivado HLS reports:

```
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch GCC as the compiler.
make[2]: Entering directory `/home/piccolboni/Qualcomm/esp/accelerators/vivado_hls/du/hls-work-virtex7/du_dma32_w8/du_acc/csim/build'
   Compiling ../../../../../tb/tb.cc in debug mode
   Compiling ../../../../../src/espacc.cc in debug mode
   Generating csim.exe
make[2]: Leaving directory `/home/piccolboni/Qualcomm/esp/accelerators/vivado_hls/du/hls-work-virtex7/du_dma32_w8/du_acc/csim/build'
--version: No such file or directory
****start*****
Test FAILED with 15 errors.
INFO: [SIM 211-1] CSim done with 0 errors.
INFO: [SIM 211-3] *************** CSIM finish ***************
```

If you change the default value of the register `in_bytes` from 16 to 0 it works. If you change the chunking factor to 1 it works as well. So, it might have something to do with the default value of the register and the chunking factor.

paulmnt, 2020-05-04:
The chunking factor of 1024 is not a divisor of 16. When you select the SystemC flow with Stratus HLS, the accelerator template is generated to handle these irregular transactions, where the size of the PLM is not an exact divisor of the input size. This is not the case, instead, for the Vivado HLS template.

The problem is therefore not the script, but the template of the accelerator for Vivado HLS.
We plan to improve the template, but fo for the time being, users who need to specify complex sequences of DMA transfers, or irregular transactions are encouraged to select the SystemC flow.

paulmnt, 2020-05-04:
One more thing: the script generates a skeleton that the user is supposed to complete by specifying the body of the computation and tuning load and store blocks to the specific needs of the target application.

Users should expect that the output of the script is HLS-ready code that describes an ESP accelerator with user-defined configuration parameters. The accelerator code, however, must be completed before running a simulation (e.g. [multiply-and-accumulate](https://esp.cs.columbia.edu/docs/cpp_acc/)).