sld-columbia/esp #44: Vivado HLS IPs (xci) are not installed automatically

paulmnt, 2020-05-11:
Vivado HLS may generate Xilixn IP definition files (_.xci_) as part of the generated RTL.
Currently, ESP installs only RTL and data (_.dat_) files into the technology-dependent folder.

Users can work around this issue by adding manually the _xci_ file to the list of source files for synthesis, but ESP should pick these files automatically with the Make target `<accelerator>-hls` and `make vivado/srcs.tcl`

Simulation of Vivado HLS IPs that include _xci_ files is not supported. There are two ways to fix simulation:
 - `make sim` target compiles all Xilinx IPs; currently the target compiles only primitives from Xilinx libraries.
 - `make sim` target generates the Xilinx IPs from _xci_ and compiles the generated output RTL files.

davide-giri, 2020-05-18:
The following patch enables this feature: xci files generated by Vivado are automatically added to the Vivado project when generating an FPGA bistream. We will close the issue once the patch has been tested and pushed.
```
--- a/utils/Makefile
+++ b/utils/Makefile
@@ -822,6 +822,8 @@ VLOG_RTL_SRCS  += $(shell (find $(ESP_ROOT)/tech/$(TECHLIB) -name "*.v" ))

 SVLOG_RTL_SRCS  = $(shell (find $(ESP_ROOT)/rtl/src -name "*.sv" ))

+IP_XCI_SRCS  = $(shell (find $(ESP_ROOT)/tech/$(TECHLIB) -name "*.xci" ))
+
 VHDL_SIM_PKGS  = $(shell (find $(ESP_ROOT)/sim/include -name "*.vhd" ))
 VHDL_SIM_SRCS  = $(shell (find $(ESP_ROOT)/sim/src -name "*.vhd" ))
 VLOG_SIM_SRCS  = $(shell (find $(ESP_ROOT)/sim/src -name "*.v" ))
@@ -1384,6 +1386,12 @@ endif
        @echo "source ./srcs.tcl" >> $@
 ifneq ("$(PROTOBOARD)","")
        @echo "set_property board_part $(PROTOBOARD) [current_project]"  >> $@
+endif
+ifneq ($(IP_XCI_SRCS),)
+       @for rtl in $(IP_XCI_SRCS); do \
+               echo "import_ip -files $$rtl" >> $@; \
+       done;
+       @echo "upgrade_ip [get_ips -all]" >> $@
 endif
        @if test -r $(ESP_ROOT)/constraints/$(BOARD)/mig.xci; then \
                echo $(SPACES)"INFO including MIG IP"; \

```

davide-giri, 2020-09-24:
Issue solved by PR #63 with commit 90c8e91.

closed 2020-09-24 by davide-giri