sld-columbia/esp #45: You should explore using IBM's OpenPOWER architecture for your chips

mfcoeid, 2020-05-12:
You should explore using IBM's OpenPOWER architecture for your chips. The Power ISA has been open-sourced, and you should consider using it for your chips. The OpenPOWER ISA is the most high performance processor architecture and ecosystem in the industry. The OpenPOWER ISA is used by the top supercomputers in the world, as well as all the big banks. It was also used by Apple's Power Macs, and the PS3, Xbox, Wii U, and more. It is easily scalable from mobile phones to the world's fastest supercomputers. Companies in the OpenPOWER foundation include Google, IBM, Samsung, Wistron, Inspur, Nvidia, NEC, Micron, ZTE, and many more. Please explore it and consider using it as your chip architecture.

paulmnt, 2020-05-12:
Thank you for the suggestion. This is a major task, but we will certainly explore the option and see if we can allocate resources to it. Do you have a particular core design to recommend that implements the OpenPOWER ISA?

seldridge, 2020-05-13:
There are only two extant implementations of open source OpenPOWER cores that I'm aware of:
  - [microwatt](https://github.com/antonblanchard/microwatt)
  - [chiselwatt](https://github.com/antonblanchard/chiselwatt)