sld-columbia/esp #50: it stops at here

CristinaZN, 2020-07-24:
**Describe the bug**
it stops at here

**To Reproduce**
Steps to reproduce the behavior:
source /home/esp/socs/xilinx-vc707-xc7vx485t/vivado/.Xil/Vivado-25650-acs30/webtalk/labtool_webtalk.tcl -notrace
 INFO: [Common 17-186] '/home/esp/socs/xilinx-vc707-xc7vx485t/vivado/.Xil/Vivado-25650-acs30/webtalk/usage_statistics_ext_labtool.xml' has been successfully sent to Xilinx on Fri Jul 24 02:11:00 2020. For additional details about this file, please refer to the WebTalk help file at /home/tools/Xilinx/Vivado/2019.2/doc/webtalk_introduction.html.
 INFO: [Common 17-206] Exiting Webtalk at Fri Jul 24 02:11:00 2020...
     INFO Waiting for DDR calibration...
     OBJCP ram.srec
     OBJCP systest.bin
 ESPLink address 192.168.1.7:46390

**Expected behavior**
i want to boot linux

**Screenshots**

[image]

davide-giri, 2020-07-25:
Hello,

The ESPLink address 192.168.1.7 seems correct, because it's the default Ethernet IP for the ESP debug unit for the FPGA board that you are using.

Looking at the _Debug link configuration_ section in the guide at this link https://www.esp.cs.columbia.edu/docs/singlecore/singlecore-guide/, are you using strategy a) or b)?

If you are using strategy a), remember to run the `ifconfig` command as described in the guide. If you are using strategy b) make sure the IP address `192.168.1.7` is in the subnet of IP addresses assigned by your router as described in the guide. Let us know if you need any clarification related to the guide.

What are the steps you did to setup a) or b)?

Thank you.
Davide

CristinaZN, 2020-07-27:
Thank you for the advice, the problem is fixed. But there is another problem when i boot linux:

Disabling lock debugging due to kernel taint                                    
Caller[f0108428]: __register_sysctl_table+0x35c/0x514                           
Caller[f04f5eec]: ipv4_sysctl_init_net+0x14/0x68                                
Caller[f02e0ab8]: ops_init.constprop.7+0xac/0x148                               
Caller[f02e0c04]: register_pernet_operations+0xb0/0x104                         
Caller[f02e0c70]: register_pernet_subsys+0x18/0x34                              
Caller[f04f5eb4]: sysctl_ipv4_init+0x34/0x58                                    
Caller[f0007128]: do_one_initcall+0x2c/0x174                                    
Caller[f04df01c]: kernel_init_freeable+0x124/0x1c0                              
Caller[f03e0b14]: kernel_init+0x4/0x104                                         
Caller[f0008234]: ret_from_kernel_thread+0xc/0x38                               
Caller[00000000]:   (null)                                                      
Instruction DUMP: 80886001                                                      
 12800076  01000000                                                             
< c6006004> 88106001                                                             
 80a0c002  0280002c                                                             
 8e100001  80a0e000                                                             
                                                                                
Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000009         
                                                                                
Press Stop-A (L1-A) to return to the boot prom                                  
---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000009

davide-giri, 2020-07-27:
Nice, I'm glad the previous problem is fixed.

Regarding the new problem could you share all the prints on the terminal starting from the beginning on the boot? That way it will be easier to understand the cause of this.

Does this happen every time you boot or is it sporadic?

My first guess is that the Linux root file system is incomplete. Can you show me the content of the `sysroot` folder in your design folder `socs/xilinx-vc707-xc7vx485t`? If you run `ls` in the `sysroot` folder and you are using the Ariane core (for the Leon3 it is pretty much the same) you should expect something like this:
```bash
applications/  bin/  dev/  etc/  init*  lib/  lib64@  linuxrc@  media/  mnt/  opt/  proc/  root/  run/  sbin/  sys/  tmp/  usr/  var
```

Thanks

CristinaZN, 2020-07-30:
here is the content of `sysroot`
[@ sysroot]$ ls
 applications  dev  init  lib32    media  opt   root  sbin  tmp  var
 bin           etc  lib   linuxrc  mnt    proc  run   sys   usr

here is full log:
?PROMLIB: Sun Boot Prom Version 0 Revision 0
 Linux version 4.9.112-g5a7ba4b-dirty (@) (gcc version 5.5.0 (Buildro0
 bootconsole [earlyprom0] enabled
 ARCH: LEON
 TYPE: Leon3 System-on-a-Chip
 Ethernet address: 00:00:c1:75:f4:b3
 CACHE: 4-way associative cache, set size 4k
 255MB HIGHMEM available.
 OF stdout device is: /a::a
 PROM: Built device tree with 43255 bytes of memory.
 Booting Linux...
 Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 112020
 Kernel command line: console=ttyS0,38400 init=/sbin/init 
 PID hash table entries: 1024 (order: 0, 4096 bytes)
 Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)                 
 Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)                   
 Sorting __ex_table...                                                           
 Memory: 440616K/449616K available (3971K kernel code, 153K rwdata, 828K rodata,)
 NR_IRQS:64                                                                      
 Console: colour dummy device 80x25                                              
 console [ttyS0] enabled                                                         
 console [ttyS0] enabled                                                         
 bootconsole [earlyprom0] disabled                                               
 bootconsole [earlyprom0] disabled                                               
 clocksource: timer_cs: mask: 0xffffffffffffffff max_cycles: 0x1d854df40, max_ids
 Calibrating delay loop... 49.35 BogoMIPS (lpj=246784)                           
 pid_max: default: 32768 minimum: 301                                            
 Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)                     
 Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)                
 devtmpfs: initialized                                                           
 OF: Duplicate name in ambapp0, renamed to "GAISLER_APBUART:1:12@2,80000100#1"   
 OF: Duplicate name in ambapp0, renamed to "GAISLER_IRQMP:1:13@0,80000200#1"     
 OF: Duplicate name in ambapp0, renamed to "GAISLER_GPTIMER:1:17@8,80000300#1"   
 OF: Duplicate name in ambapp0, renamed to "eb_020:235:32@3,8000d900#1"          
 OF: Duplicate name in ambapp0, renamed to "GAISLER_ETHMAC:1:29@c,80080000#1"    
 OF: Duplicate name in ambapp0, renamed to "01_092:1:146@b,80001000#1"           
 OF: Duplicate name in ambapp0, renamed to "eb_021:235:33@4,8000e000#1"          
 OF: Duplicate name in ambapp0, renamed to "eb_013:235:19@3,80011400#1"          
 OF: Duplicate name in ambapp0, renamed to "GAISLER_APBUART:1:12@2,80000100#2"   
 OF: Duplicate name in ambapp0, renamed to "GAISLER_IRQMP:1:13@0,80000200#2"     
 OF: Duplicate name in ambapp0, renamed to "GAISLER_GPTIMER:1:17@8,80000300#2"   
 OF: Duplicate name in ambapp0, renamed to "eb_020:235:32@3,8000d900#2"          
 OF: Duplicate name in ambapp0, renamed to "GAISLER_ETHMAC:1:29@c,80080000#2"    
 OF: Duplicate name in ambapp0, renamed to "01_092:1:146@b,80001000#2"           
 OF: Duplicate name in ambapp0, renamed to "eb_021:235:33@4,8000e000#2"          
 OF: Duplicate name in ambapp0, renamed to "eb_013:235:19@3,80011400#2"          
 OF: Duplicate name in ambapp0, renamed to "GAISLER_APBUART:1:12@2,80000100#3"   
 OF: Duplicate name in ambapp0, renamed to "GAISLER_IRQMP:1:13@0,80000200#3"     
 OF: Duplicate name in ambapp0, renamed to "GAISLER_GPTIMER:1:17@8,80000300#3"   
 OF: Duplicate name in ambapp0, renamed to "eb_020:235:32@3,8000d900#3"          
 OF: Duplicate name in ambapp0, renamed to "GAISLER_ETHMAC:1:29@c,80080000#3"    
 OF: Duplicate name in ambapp0, renamed to "01_092:1:146@b,80001000#3"           
 OF: Duplicate name in ambapp0, renamed to "eb_021:235:33@4,8000e000#3"          
 OF: Duplicate name in ambapp0, renamed to "eb_013:235:19@3,80011400#3"          
 clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 191s
 futex hash table entries: 256 (order: -1, 3072 bytes)                           
 NET: Registered protocol family 16                                              
 SCSI subsystem initialized                                                      
 clocksource: Switched to clocksource timer_cs                                   
 FS-Cache: Loaded                                                                
 CacheFiles: Loaded                                                              
 NET: Registered protocol family 2                                               
 TCP established hash table entries: 2048 (order: 1, 8192 bytes)                 
 TCP bind hash table entries: 2048 (order: 1, 8192 bytes)                        
[... 129 lines cut ...]
 Caller[f0108428]: __register_sysctl_table+0x35c/0x514                           
 Caller[f04f5eec]: ipv4_sysctl_init_net+0x14/0x68                                
 Caller[f02e0ab8]: ops_init.constprop.7+0xac/0x148                               
 Caller[f02e0c04]: register_pernet_operations+0xb0/0x104                         
 Caller[f02e0c70]: register_pernet_subsys+0x18/0x34                              
 Caller[f04f5eb4]: sysctl_ipv4_init+0x34/0x58                                    
 Caller[f0007128]: do_one_initcall+0x2c/0x174                                    
[... cut ...]
 Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000009         
                                                                                 
 Press Stop-A (L1-A) to return to the boot prom                                  
 ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000009
                                                                                 
 random: crng init done                                                          

Thanks

paulmnt, 2020-07-30:
Ethernet Debug Link problem is solved. Moving the conversation to #57

closed 2020-07-30 by paulmnt

LucaUrbinati44, 2022-07-25:
Hello. I have the same initial problem of @CristinaZN, that is the baremetal stuck at `ESPLink address`.
I'm running a SoC with no accelerators, just one processor (Arianne), two memory tiles and one I/O.

I think that my problem is in the physical connection between host PC and my FPGA, which is a profpga xc7v2000t. I have the [Ethernet module mounted](url) on the FPGA connected to the host PC with a normal Ethernet cable, i.e. a cable without [crossover](https://www.computercablestore.com/what-are-ethernet-crossover-cables). Could this be the problem?

The host PC has two network cards: one is used to connect it to a LAN and to the internet (called "enp0s31f6", 172.168.x.y), while the second is only used for the connection to the FPGA (called "p4p1", 192.168.1.3).

I set 192.168.1.2 in the grlib-xconfig GUI, following the [single core guide](https://esp.cs.columbia.edu/docs/singlecore/singlecore-guide/#a-direct-link).

Moreover, the profpga [quad motherboard](https://www.profpga.com/products/motherboards-overview/profpga-quad) is connected to the same LAN of the PC and has an IP such as 172.168.x.z. 

When the profpga is down, I see the interface "p4p1" up by issuing `ifconfig` on the terminal. Instead, after the bitstream is sent to the FPGA (`make fpga-program`), I see interface "p4p1" going to "unconnected" state from the Network Manager of Linux GUI. Therefore, when running `make fpga-run`, since the interface "p4p1" is down, the program stuck at `ESPLink address`. Even when I turn off the profpga (`make profpga-close-fpga`), I am not able to turn back up interface "p4p1" of the host (I have to remove it and create a new interface).

[image]  

[image]

Should I use a crossover Ethernet cable to make the direct link?
Should I try to use a switch instead of a direct link?
Did I make any mistake in the IP address configuration of host PC or profpga?

Any help is appreciated.
Thank you