sld-columbia/esp #65: Support Ariane w/ PMP

paulmnt, 2020-10-01:
If bumping Ariane to include PMP support, we see a kernel panic when attempting to run init.

Moving discussion from  #61

jctullos, 2020-10-04:
@paulmnt 

So one thing I just had happen to me on another build, I had almost a similar error that we both saw in regards to the Init panic at the end. It's not exactly the same, but on my new error it references the same file: ld-2.26.so. After looking around, I realized the issue for this other processor was that I had built it for RV64IMAFDC, but in the DTS I passed RV64IMAC. There was a fix that someone referenced in: https://github.com/riscv/riscv-pk/issues/166 that said the same thing, if they pass the wrong ISA to BBL/Linux, it causes the same issue. 

I tried to look around at where ESP actually specifies the ISA when building the BBL/Linux, but I couldn't find anything concrete. Just a thought, it might be worth a look at the bootrom and see if passing just RV64IMAC doesn't have the same error. Maybe something isn't getting passed around there.

davide-giri, 2020-10-04:
This is a good lead, thank you. Let me see if I find where the ISA gets when building BBL/Linux.

 One place where the ISA definitely gets specified is the device tree. This is the line of Python specifying the ISA in the device tree.
```bash
utils/socmap/socmap_gen.py:    fp.write("      riscv,isa = \"rv64imafdc\";\n")
```

jctullos, 2020-10-04:
Yeah, that was all I found. The ISA in the bbl build sets to a default value. I'm in progress of reprogramming my VCU118 back with the ESP build, and I've changed the DTS to rv64IMAC, I'll then see if it succeeds. It fixed the other processor that had the same problem.

Kendidi, 2020-10-05:
I wonder when we build code for FPGAs like VCU128, which of the following variants will be used?

    esp/third-party/ariane/
    esp/third-party/ariane/fpga/
    esp/third-party/ariane/openpiton/
    esp/soft/ariane/

paulmnt, 2020-10-05:
@Kendidi we should try to keep issues separate by topic. I am replying to you on #66 but I can't mention you there, because you haven't posted on that issue yet.

Kendidi, 2020-10-05:
Thank you @paulmnt !  I wasn't sure where to post this at first.  I apologize!