sld-columbia/esp #67: Error in synthesis step of "How to: design a single-core SoC"

JuanEsco063, 2020-10-09:
FPGA prototype step of guide "How to: design a single-core SoC" fails when trying to synthesize mig and sgmii

To reproduce, follow the steps in the How to: design a single-core SoC" guide until the FPGA pretty step.
1) type "make vivado-syn" 

I expected to be able to complete the example without any issues.

See screenshot of errors:

[image]

 - OS: Ubuntu 18.04.5
 - CAD tools versions: Vivado and Vitis 2019.2. No RTL simulation CAD tools. Latest version of all other software packets.

davide-giri, 2020-10-09:
Hi @JuanEsco063,

Could you share the `vivado_syn.log` file generated in the folder where you launched `make vivado-syn`? The OS and Vivado versions you are using are fine.

Thank you

JuanEsco063, 2020-10-09:
Hi Davide,

Here is the log file

[vivado_syn.log](https://github.com/sld-columbia/esp/files/5356358/vivado_syn.log)

davide-giri, 2020-10-09:
Could you also share these two log files:
`mig_synth_1: /home/juan/esp/socs/xilinx-vc707-xc7vx485t/vivado/esp-xilinx-vc707-xc7vx485t.runs/mig_synth_1/runme.log`
`sgmii_synth_1: /home/juan/esp/socs/xilinx-vc707-xc7vx485t/vivado/esp-xilinx-vc707-xc7vx485t.runs/sgmii_synth_1/runme.log`

Thanks!

JuanEsco063, 2020-10-09:
It seems a problem with licensing. Thank you for your help! I will contact IT since I know we have a full license.

JuanEsco063, 2020-10-12:
After acquiring the license, the synthesis still fails. Attached are the requested files. A quick google search of the error why the synthesis fails led me to this article.

https://forums.xilinx.com/t5/Synthesis/Synthesis-process-died-without-any-design-errors-reported/td-p/871322

[vivado_syn.log](https://github.com/sld-columbia/esp/files/5366333/vivado_syn.log)
[runme[1].log](https://github.com/sld-columbia/esp/files/5366335/runme.1.log)
[runme.log](https://github.com/sld-columbia/esp/files/5366336/runme.log)

davide-giri, 2020-10-12:
It's possible that your system ran out of memory. How much RAM do you have? If you run `dmesg` right after the failure there may be an "out of memory" message to confirm this hypothesis. One idea to reduce the memory requirements of Vivado is to edit `vivado/syn.tcl` in your design folder: wherever it says `-jobs 12` you can specify a smaller number.

davide-giri, 2020-10-12:
I also suggest to try one more time, to see if it fails in the same way.

JuanEsco063, 2020-10-12:
I increased the RAM amount to 8 GB on my VM and lowered the thread count to 4 and the single core example completed successfully. I am trying to run the NVDLA example right now. If it finishes successfully, I will close the issue. Thank you for your help!

JuanEsco063, 2020-10-12:
The example failed but ran the dmesg command and indeed it is an out of memory error. What amount of RAM do you recommend?

davide-giri, 2020-10-12:
What part failed? If it's Vivado again remember to change the `-jobs` argument like you did earlier.

JuanEsco063, 2020-10-12:
Yes, it was Vivado. I changed the number of jobs for both synthesis and implementation to 1 and it still crashed with an out of memory error

davide-giri, 2020-10-12:
If you have more RAM available I would increase the memory size for the VM. If you don't we should look for another solution.

JuanEsco063, 2020-10-12:
Currently I am on a Mac with 16 GB of RAM. Allotting more than 8GB to the VM makes the host OS unstable

davide-giri, 2020-10-12:
I see, the problem is using Vivado on your machine. Here are some guidelines on Vivado's memory requirements for each FPGA target: https://www.xilinx.com/products/design-tools/vivado/memory.html. Which FPGA are you targeting?

JuanEsco063, 2020-10-12:
For the NVDLA accelerator it is targeting the xcvu9p which from the table it seems it requires  20-32 GB which explains the out of memory error.

davide-giri, 2020-10-12:
Are you targeting the VCU118 board because you have it? Have you used Vivado for that board before? I see that in the picture you shared in the first message you were targeting the Xilinx VC707 board. If you have the VC707 board, I suggest to target that one, since it requires less memory and you already succeeded with it.

closed 2020-10-29 by JuanEsco063