sld-columbia/esp #88: Can I use nvdla accelerator in docker？

MrHaiCoder, 2021-02-01:


davide-giri, 2021-02-01:
Hi,

Yes, you can. You can pretty much use ESP in full in a Docker container.

The instructions for using the ESP Docker are in the [Docker section of the ESP Setup Guide](https://www.esp.cs.columbia.edu/docs/setup/setup-guide/#docker). The idea for using 

The instructions for using NVDLA with ESP are in the [How to: integrate a third-party accelerator (e.g. NVDLA)](https://www.esp.cs.columbia.edu/docs/thirdparty_acc/).

Let us know if you have any questions!

MrHaiCoder, 2021-02-02:
I am using centos7 full image to run make ESP xconfig according to how to: integrate a third party accelerator (e.g. nvdla), but an error is reported. Do I need to install vivado in docker?

[image]

davide-giri, 2021-02-02:
Did you follow the instructions for using CAD tools ([here](https://www.esp.cs.columbia.edu/docs/setup/setup-guide/#cad-tools-with-docker)) and setting environment variables ([here](https://www.esp.cs.columbia.edu/docs/setup/setup-guide/#environment-variables-with-docker)) inside the Docker? That's in the setup guide I mentioned in the previous message. Let me know if you still have issues or questions after reading those sections of the guide.

MrHaiCoder, 2021-02-04:
> 您是否遵循了在Docker内部使用CAD工具（[此处](https://www.esp.cs.columbia.edu/docs/setup/setup-guide/#cad-tools-with-docker)）和设置环境变量（[此处](https://www.esp.cs.columbia.edu/docs/setup/setup-guide/#environment-variables-with-docker)）的说明？这就是我在上一条消息中提到的设置指南中。阅读指南中的这些部分后，如果您仍有问题或疑问，请告诉我。

thank you

davide-giri, 2021-02-13:
@MrHaiCoder did you manage to use the Docker? Do you need additional assistance or should we close this issue?

Thank you

closed 2021-02-21 by davide-giri