sld-columbia/esp #100: Single-core tutorial: Can not launch Linux on vcu118

andy39866821, 2021-04-07:
I am trying to run **make fpga-run-linux** in single core design tutorial, but it failed when mounting root filesystem.
Here is the message in minicom when launching Linux.
```
Linux version 4.9.112-g5a7ba4b (ycchung@ic13.cs.nthu.edu.tw) (gcc version 5.5.0 (B3 System-on-a-Chip
Ethernet address: 00:00:5d:17:4b:b9
CACHE: 4-way associative cache, set size 4k
255MB HIGHMEM available.
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 112671
Kernel command line: console=ttyS0,38400 init=/sbiDentry cache hash table entries: 32768 (order: 5, 131072 bytes)
Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
Memory: 443256K/452220K available (3971K kernel code, 157K rwdata, 832K rodata, 1376K init, 129K bss, 8964K reserved, 5
console [ttyS0] enabled
console [ttyS0] enabled
bootconsoCalibrating delay loop... Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
devtmpfs: initialized
clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
futex hash table entries: 256 (order: -1, 3072 bytes)
clocksource: Switched to clocksource timer_cs
CacheFiles: Loaded
TCP established hash table entries: 2048 (order: 1, 8192 bytes)
TCP: Hash tables configured (established 2048 bind 2048)
UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
RPC: Registered named UNIX socket transport module.
RPC: Registered tcp transport module.
random: fast init done
FS-Cache: Netfs 'nfs' registered for caching
NFS: Registering the id_resolver key type
Key type id_resolver registered
Key type id_legacy registered
bounce: pool size: 64 pages
io scheduler deadline registered
Serial: GRLIB APBUART driver
grlib-apbuart at 0x80000100, irq 2
loop: module loaded
sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
Key type dns_resolver registered
VFS: Cannot open root device "(null)" or unknown-block(0,0): error -6
0100            4096 ram0  (driver?)
0102            4096 ram2  (driver?)
0104            4096 ram4  (driver?)
0106            4096 ram6  (driver?)
0108            4096 ram8  (driver?)
010a            4096 ram10  (driver?)
010c            4096 ram12  (driver?)
010d            4096 ram13  (driver?)
010e            4096 ram14  (driver?)
010f            4096 ram15  (driver?)
Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
CPU: 0 PID: 1 Comm: swapper Not tai : kernel_init_freeable+0x1b0/0x1c0 ]
[f03e0b14 : kernel_init+0x4/0x104 ]
[f0008234 : ret_from_kernel_thread+0xc/0x38 ]
[nt root fs on unknown-block(0,0)

 CTRL-A Z for help | 38400 8N1 | NOR | Minicom 2.6.2  | VT102 |    
```

Everything in the previous step is fine, **make fpga-run** can generate "Hello ESP" in minicom as tutorial said.
Here is my hardware and setting.
Is there any wrong with my config setting?
```
OS: CentOS7
FPGA: Xilinx vcu118

esp config: leon3, not used cache, other are default value.

Makefile setting in esp/socs/xilinx-vcu118-xcvu9p/ :
### Xilinx Vivado hw_server ###
FPGA_HOST ?= localhost
XIL_HW_SERVER_PORT ?= 3121
# IP address or host name of the host connected to the FPGA
UART_IP ?=
UART_PORT ?=
```

paulmnt, 2021-04-07:
hi @andy39866821 

This error message indicates that you did not build correctly the template for the root file system. This could happen when some compilation step of buildroot for RISC-V fail due to missing system packages or libraries.

Please run again the toolchain scripts, skip the steps for baremetal and Linux compilers (GCC), but repeat the step for sysroot. See the [setup guide](https://esp.cs.columbia.edu/docs/setup/setup-guide/#software-toolchain) for details.
  
Look carefully at the messages on the shell while buildroot executes and make sure you have installed all [required packages](https://esp.cs.columbia.edu/docs/setup/setup-guide/#software-packets)
 
Once the sysroot template has been rebuilt correctly, enter the design folder for `VCU118` and run `make linux-distclean` to remove the incomplete sysroot folder. Next run `make linux -j 8` again to recompile Linux with the correct root file system embedded.

Please let us know if you have additional issues.

andy39866821, 2021-04-08:
@paulmnt 
It's work. After reinstall the toolchains and recompile Linux, now I can login the Linux on FPGA as root like the tutorial said.
Thanks for your helping!

closed 2021-04-13 by paulmnt