sld-columbia/esp #102: FPGA prototyping with Ariane core has "Test FAILED" issue ("fpga-run-linux", vcu128 board)

AltiumHanChou, 2021-04-13:
**Describe the bug**

>   I use vcu128 and test this [tutorial](https://www.esp.cs.columbia.edu/docs/systemc_acc/systemc_acc-guide/).
>   In the section "FPGA prototyping"  has "Test FAILED" issue when I run the command "fpga-run-linux".
> 

**Expected behavior**

>   I got validation pass information when I run the command `TEST_PROGRAM=./soft-build/<cpu>/baremetal/mac_stratus.exe make fpga-run`
>   
>   log:
> 
> 
> ```
> ESP-Ariane first-stage boot loader                                              
>                                                                                 
> Scanning device tree...                                                         
> [probe] sld,mac_stratus.0 registered                                            
>         Address   : 0x60010000                                                  
>         Interrupt : 6                                                           
> **************** sld,mac_stratus.0 ****************                             
>   memory buffer base-address = 0xa0100cd0                                       
>   ptable = 0xa0107280                                                           
>   nchunk = 1                                                                    
>   --------------------                                                          
>   Generate input...                                                             
>   -> Non-coherent DMA                                                           
>   Start...                                                                      
>   Done                                                                          
>   validating...                                                                 
>   ... PASS      
> ```
>
> But I run the command `TEST_PROGRAM=./soft-build/<cpu>/baremetal/mac_stratus.exe make fpga-run-linux`.
>   I got "Test FAILED" when I run the program ./mac_stratus.exe.
> 
> ```
> ====== mac_stratus.0 ======                                                     
>                                                                                 
>   .mac_n = 1                                                                    
>   .mac_vec = 100                                                                
>   .mac_len = 64                                                                 
>                                                                                 
>   ** START **                                                                   
>   > Test time: 10188053 ns                                                      
>     - mac_stratus.0 time: 888373 ns                                             
>                                                                                 
>   ** DONE **                                                                    
> + Test FAILED                                                                   
>                                                                                 
> ====== mac_stratus.0 ====== 
> ```

**Desktop (please complete the following information):**

>  - OS: Ubuntu 18.04
>  - CAD tools versions: Xilinx Vivado 2019.2, STRATUS_19.12.100

**Additional context**

> Because It pass without run the linux on FPGA, I assume it could be something go wrong when I run `make linux`. ( I am look into the part which about  `make linux` . If I fix something I will update here)

paulmnt, 2021-04-13:
hi @AltiumHanChou 

When you run linux, there's no need to specify the TEST_PROGRAM. For that target, in fact, the main program that executes is Linux itself.

Since you have correct execution on FPGA, it's unlikely that you have a hardware bug. I would check the Linux application code. The reason for failure could be wrong computation of the golden output, or wrong configuration of the accelerator.

The Linux application should be located at the path `eps/accelerators/stratus_hls/mac_stratus/sw/linux/app`.

If you can't spot the issue, I would recommend printing some of the output values, as well as the expected golden output.
The result of that print should help you debug.

Remember, every time you change the application, to recompile it and update the Linux root file system with the following make targets before rebooting.

```
make mac_stratus-app
make linux
```
Alternatively, you can simply recompile the app with `make mac_stratus-app` and use scp to copy it into the running ESP instance. If the problem is just a software bug, in fact, you most likely don't need to reboot to test your changes to the application.
The compiled application is copied into `esp/socs/<design>/soft-build/ariane/sysroot/applications/test/mac_stratus.exe` and the IP of the ESP instance can be obtained using `ifconfig` from the UART interface.

davide-giri, 2021-05-26:
Hi @AltiumHanChou, were you able to get a correct validation of the `mac` accelerator Linux application? Has this issue been resolved?

Thanks!

AltiumHanChou, 2021-05-26:
Hi ~

Yes. I think all of the issue has been resolved. All of test in new release
is pass.
Now I am trying to make and test some different “mac” in ESP platform .
Thank you for your help. That was very helpful to me to do these research.

Altium han chou

davide-giri, 2021-05-26:
That's good to hear! Please don't hesitate to contact us with any further question, either here or by email.

closed 2021-05-26 by davide-giri