sld-columbia/esp #107: 'make linux' creates a single-core image ?

rdolbeau, 2021-04-19:
**Describe the bug**

I've followed the tutorial for multicore with leon3; everything went fine and I can log on the SoC via the console.
However, Linux only uses one core out of 4.

**To Reproduce**

Steps to reproduce the behavior:
1. Follow the multicore tutorial

**Expected behavior**

I expected to see 4 cores in Linux, but there's only one.

This is not surprising, as in the linux configuration ./soft-build/leon3/linux-build/.config I see:

CONFIG_BROKEN_ON_SMP=y
CONFIG_GENERIC_SMP_IDLE_THREAD=y
# CONFIG_SMP is not set

**Screenshots**

N/A

**Desktop (please complete the following information):**

Docker image with recommended Vivado (2019.2) installed in it, bitstream & software generation were fine.

**Additional context**

Great tutorial thanks :-)

davide-giri, 2021-04-19:
Hi,

In the local Makefile (the one in the folder where you are launching all make commands) you should uncomment this line
```
LINUX_SMP = 1
```

Then you need to recompile linux:
```
make linux-distclean
make linux
```

Let me know if this solves it. Thanks!

rdolbeau, 2021-04-19:
@davide-giri Thanks for the quick answer :-) Just tested, indeed it works fine that way, I have 4 Leon3 now in Linux. Maybe this should be mentioned in the multicore tutorial?

davide-giri, 2021-04-19:
Yes, it definitely should be mentioned in the tutorial. Thanks for catching this, I'll make sure to update the tutorial asap.

Thanks!

closed 2021-04-19 by davide-giri