sld-columbia/esp #110: Temp build folder for toolchain compilation conflict on shared computer

paulmnt, 2021-04-26:
**Is your feature request related to a problem? Please describe.**
On a shared computer, multiple users cannot build the RISC-V toolchain because of a conflict with the temporary folder `/tmp/_riscv_build`.

**Describe the solution you'd like**
Generate a different temporary folder when running the script to resolve the conflict.

**Describe alternatives you've considered**
Have the script delete the temporary folder when it exists. This solution could work, however, it would prevent concurrent executions and it may fail in case a step of the script exists with an error due to missing system packages.

davide-giri, 2021-04-26:
How about going back to having the temporary `_riscv_build` and `_leon3_build` folders local in each user's home folder?

paulmnt, 2021-04-26:
that's an option of course.

davide-giri, 2021-05-27:
Fixed with PR #119.

closed 2022-03-08 by jzuckerman