sld-columbia/esp #136: ESP simple configuration with leon3 and no caches problem

yyhanafy, 2021-12-19:
Hi,

  Just installed ESP with Xilinx and Modelsim, and tried the simplest configuration with Leon3 and no caches. I skipped the make llc-hls and l2c-hls as indicated in the tutorial, since I am not using an L2 or an LLC (I did that because I do NOT have a license for stratus, only Vivado and Modelsim!!). When I tried to simulate, I got the following error

root@b4816a239ac5:/home/espuser/esp/socs/xilinx-vc707-xc7vx485t$ make sim
    DIFF checking .esp_config...
    INFO Using custom configuration found in ".esp_config" for ESP

Generating ESP configuration...
Created global constants definition into 'esp_global.vhd'
Created configuration into 'socmap.vhd'
Created ESPLink header into 'socmap.h'
Created RTL caches configuration into 'cache_cfg.svh'
Created kernel module load script into 'S64esp'
Created configuration into 'power.h'
Created configuration into 'mmi64_regs.h'
    MKDIR socketgen
    RUN socketgen
    INFO: No accelerators found in /home/espuser/esp/tech/virtex7/acc.
          Please run 'make accelerators' or make <accelerator>-hls.
          Get available accelerators with 'make print-available-accelerators'
    WARNING: No caches found in /home/espuser/esp/tech/virtex7/sccs.
             Please check the "Use RTL" option in the "Cache Configuration" tab when configuring ESP.
    INFO: Retrieving information for NV_NVDLA
          NV_NVDLA: NVIDIA, NVDLA small, ID 100
    INFO: Generating RTL to ./socketgen
    MKDIR /home/espuser/esp/.cache/modelsim/xilinx_lib
/bin/sh: 3: vivado: not found
 ERROR: Xilinx library compilation failed!
/home/espuser/esp/utils/make/modelsim.mk:36: recipe for target '/home/espuser/esp/.cache/modelsim/xilinx_lib' failed
make: *** [/home/espuser/esp/.cache/modelsim/xilinx_lib] Error 1
root@b4816a239ac5:/home/espuser/esp/socs/xilinx-vc707-xc7vx485t$**Describe the bug**

I do appreciate any help
--Y

biruk-belay, 2021-12-19:
Hi,

I see an error related to the Vivado installation path. Did you set the
vivado installation path in the current shell ?

With Regards

- biRuk

> Hi,
>
> Just installed ESP with Xilinx and Modelsim, and tried the simplest
> configuration with Leon3 and no caches. I skipped the make llc-hls and
> l2c-hls as indicated in the tutorial, since I am not using an L2 or an LLC
> (I did that because I do NOT have a license for stratus, only Vivado and
> Modelsim!!). When I tried to simulate, I got the following error
>
> DIFF checking .esp_config...
> INFO Using custom configuration found in ".esp_config" for ESP
>
> Generating ESP configuration...
> Created global constants definition into 'esp_global.vhd'
> Created configuration into 'socmap.vhd'
> Created ESPLink header into 'socmap.h'
> Created RTL caches configuration into 'cache_cfg.svh'
> Created kernel module load script into 'S64esp'
> Created configuration into 'power.h'
> Created configuration into 'mmi64_regs.h'
> MKDIR socketgen
> RUN socketgen
> INFO: No accelerators found in /home/espuser/esp/tech/virtex7/acc.
> Please run 'make accelerators' or make -hls.
> Get available accelerators with 'make print-available-accelerators'
> WARNING: No caches found in /home/espuser/esp/tech/virtex7/sccs.
> Please check the "Use RTL" option in the "Cache Configuration" tab when
> configuring ESP.
> INFO: Retrieving information for NV_NVDLA
> NV_NVDLA: NVIDIA, NVDLA small, ID 100
> INFO: Generating RTL to ./socketgen
> MKDIR /home/espuser/esp/.cache/modelsim/xilinx_lib
> /bin/sh: 3: vivado: not found
Here is where that error is being generated.

yyhanafy, 2021-12-24:
Thank you biRuk,

 

I fixed this small issue, now to the bigger ones, is there anyway to avoid using modelsim? I am having problems with getting licenses for both modelsim, catapult and stratus. The only license I was able to get is Vivado coming with the VC709 Virtex 7 connectivity kit. My 3 questions would be

 

1.	Vivado comes with a simulator, how do we adjust the code to use it instead of modelsim?
2.	Can we switch Catapult HLS and Stratus HL with Vivado HLS and how do we do that?
3.	How do we configure ESP to use the VC709 Virtex-7 connectivity kit as the hardware platform?

 

Thank you 

--Y

klchiu, 2022-01-13:
Hi yyhanafy,
1.	Vivado comes with a simulator, how do we adjust the code to use it instead of modelsim?
--> I haven't tried the simulator comes with Vivado, maybe you can try to change the options in "project.tcl"?
2.	Can we switch Catapult HLS and Stratus HL with Vivado HLS and how do we do that?
--> Yes, please refer to the guide [How to: design an accelerator in C/C++ (Xilinx Vivado HLS)](https://esp.cs.columbia.edu/docs/cpp_acc/)
3.	How do we configure ESP to use the VC709 Virtex-7 connectivity kit as the hardware platform?
--> We don't support vc709 for now. I think you may need to check and remap the pins.

closed 2023-04-05 by jzuckerman