sld-columbia/esp #139: Is Vivado Xsim supported as a simulator by ESP (could not find a script for it!)

yyhanafy, 2022-01-13:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

klchiu, 2022-01-13:
Hi, currently it is not supported

paulmnt, 2022-01-14:
Just a little extra info on the topic: I tried using xsim in the past, but I faced a lot of issues. In the end, once I managed to get a working script, the simulation started but was not progressing.

I never investigated why so, but given that the free simulator in Vivado is quite slow for such large designs, I did not invest more time.

So, I believe it is possible to add support for this, but it won’t be as simple as adding support for l, say, VCS.

hope this helps.