sld-columbia/esp #141: ESP configuration GUI not opening:

mayank-kabra2001, 2022-01-20:
Steps - 
# Move to the Xilinx VC707 working folder
cd <esp>/socs/xilinx-vc707-xc7vx485t
make esp-xconfig

Error facing - 
Running interactive ESP configuration...
X11 connection rejected because of wrong authentication.
X11 connection rejected because of wrong authentication.
Traceback (most recent call last):
  File "/home/espuser/esp/tools/socgen/esp_creator.py", line 268, in <module>
    root = Tk()
  File "/usr/lib/python3.6/tkinter/__init__.py", line 2023, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: couldn't connect to display "localhost:10.0"
/home/espuser/esp/utils/make/esp.mk:47: recipe for target 'esp-xconfig' failed
make: *** [esp-xconfig] Error 1

Screenshot - 
[image]

Help Required !!

klchiu, 2022-01-20:
Hi it seems like a display issue. Are you using a remote server? You can test with xclock to make sure you have the correct X11 forwarding settings

mayank-kabra2001, 2022-01-20:
Yes I am using the docker on the remote server. I am able to establish X11 connection in the server but not in the docker. 
[image]

closed 2022-01-24 by mayank-kabra2001

SaltedFishLZ, 2022-09-24:
Hi, Mayank. Could you please try to `sudo bash` to switch to the root user?
I also found this issue. As root, I can open GUI applications.

SaltedFishLZ, 2022-09-24:
BTW, is it possible to run as a non-root user to show GUI applications in podman (I am using RHEL 8)? @klchiu

klchiu, 2022-11-15:
Hi @SaltedFishLZ  sorry I just read the message. You shouldn't need to be sudo user to open the GUI. I've never used podman but I can try to see what I can do in the next few weeks

TrungNgoc20172869, 2023-05-12:
> Steps -
> 
> # Move to the Xilinx VC707 working folder
> cd /socs/xilinx-vc707-xc7vx485t make esp-xconfig
> 
> Error facing - Running interactive ESP configuration... X11 connection rejected because of wrong authentication. X11 connection rejected because of wrong authentication. Traceback (most recent call last): File "/home/espuser/esp/tools/socgen/esp_creator.py", line 268, in root = Tk() File "/usr/lib/python3.6/tkinter/**init**.py", line 2023, in **init** self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use) _tkinter.TclError: couldn't connect to display "localhost:10.0" /home/espuser/esp/utils/make/esp.mk:47: recipe for target 'esp-xconfig' failed make: *** [esp-xconfig] Error 1
> 
> Screenshot - [image]
> 
> Help Required !!

Hi @klchiu and @mayank-kabra2001, I also have the problems of GUI start like this. How try to run `sudo make esp-xconfig` but it requires espuser password and I dont know what it is. Could you tell me how to fix this?

klchiu, 2023-05-13:
Hi @TrungNgoc20172869 the password for espuser should be espuser. But I think your problem is about the Xforwarding. When you start the docker container, do you set the DISPLAY variable?