sld-columbia/esp #142: make sim not running

mayank-kabra2001, 2022-01-22:
Attaching screenshot of error that rtl simulation not running.

[image]

i didnt get why we need to install vivado? It should already be there in the docker. 

Help required!!

jzuckerman, 2022-01-24:
Hello! The docker image doesn't come with any of the CAD tools installed. Please see instructions on using CAD tools with Docker here: https://esp.cs.columbia.edu/docs/setup/docker/#cad-tools-with-docker

mayank-kabra2001, 2022-01-24:
Hi , I want to install toolchains mentioned in the https://esp.cs.columbia.edu/docs/setup/setup-guide/#cad-tools. What is the password of espuser ? Am I doing something wrong ? 

[image]

mayank-kabra2001, 2022-01-24:
Got it!! its default only.. Thanks!!

mayank-kabra2001, 2022-01-25:
One more question !! If I open the docker with a volume using this command. 

 sudo docker run -it --user espuser --security-opt label=type:container_runtime_t --network=host -e DISPLAY=$DISPLAY -v my-vol:/shared-volume --name my-container-02 columbiasld/esp:ubuntu18-full /bin/bash

It will open for the first time , then after that it will show this error - 
[image]

Now if I close this container or start the new container, the toolchain that I installed is gone !! Is there any way to solve this issue ?

jzuckerman, 2022-01-25:
Once you open a docker container and exit it, you can re-open the session with 
`docker attach <container-ID>`
You can find the container-ID with:
`docker ps -a` 
Checkout other useful commands at the bottom of the Docker setup guide: https://esp.cs.columbia.edu/docs/setup/docker/

mayank-kabra2001, 2022-01-25:
Ohh got it !! Thanks !!

mayank-kabra2001, 2022-01-26:
I have install modelsim and Vivado both on my system. Now in esp_env_cad.sh when I want to change my environment variables, it works good for modelsim but not vivado. 

Changes in esp_env_cad.sh - 

[image]

Error - 

[image]

I can still source it in my system, which means that it is working correct - 

[image]

Help Required !!

mayank-kabra2001, 2022-01-26:
After resolving this sourcing issue - I am getting this error - 
[image]

Help Required!!

jzuckerman, 2022-01-26:
It looks like in your `esp_env_cad.sh`, you have the command that adds the modelsim to your PATH variable commented out (the first line under the Mentor section). You need this line for modelsim to be found.

mayank-kabra2001, 2022-01-27:
That is the license of modelsim. Do we really need that?

jzuckerman, 2022-01-27:
Looks like modelsim can't find the expected libraries from Xilinx. Please note that ESP supports Vivado version 2019.2, and I see that you are using Vivado 2020.1.

mayank-kabra2001, 2022-01-31:
Hi, 

I tried to install Vivado 2019.2 and then check, its still showing the errors. 
One question - Do we need to have licenses for Both Xilinx and modelsim ? 

Thanks.

closed 2022-02-03 by mayank-kabra2001

TrungNgoc20172869, 2023-05-12:
> Hi , I want to install toolchains mentioned in the https://esp.cs.columbia.edu/docs/setup/setup-guide/#cad-tools. What is the password of espuser ? Am I doing something wrong ?
> 
> [image]

Hi, I am facing the same issue and dont know what the password is. Could you tell me the espuser password ?