sld-columbia/esp #144: Issue while implementing single-core soc on esp

mayank-kabra2001, 2022-02-14:
I am receiving error while doing rtl simulation using {make xmsim}. 

[image]

I have used the following configuration - 
[image]

Can someone help me out!!

jzuckerman, 2022-02-14:
Hello, I just pushed a fix for this to the master branch. For Xcelium, you'll also need to apply a patch to the Ariane core as indicated here: https://esp.cs.columbia.edu/docs/setup/setup-guide/#patching-ariane-for-xcelium-simulator .

harshitaaaaaa, 2022-02-14:
Yes, I have applied this patch but I am still facing the issue. I downloaded the patch mentioned in the link ([here](https://esp.cs.columbia.edu/docs/setup/xcelium_ariane.patch))

mayank-kabra2001, 2022-02-14:
Yes, I have applied this patch but I am still facing the issue. I downloaded the patch mentioned in the link ([here](https://esp.cs.columbia.edu/docs/setup/xcelium_ariane.patch))

closed 2022-02-14 by mayank-kabra2001