sld-columbia/esp #148: Error while using hls4ml

mayank-kabra2001, 2022-02-18:
Hi, I was using hls4ml and trying to implement accelerator using this tutorial (https://www.esp.cs.columbia.edu/docs/hls4ml/hls4ml-guide/). 

In the Accelerator generation section while doing hls4ml convert -c keras-config.yml , I found some errors related to python. 
[image]

Can someone help me out!!

GiuseppeDiGuglielmo, 2022-02-20:
@mayank-kabra2001 I am assuming that you are using the latest version of hls4ml (i.e. `main` branch). Can you please try the following steps:

```
git clone https://github.com/hls-fpga-machine-learning/hls4ml.git
cd hls4ml
git checkout v0.6.0 
pip install .
```

if it does not work yet, please try:
```
git checkout v0.5.0 
pip install .
```

Let me know,
Giuseppe

mayank-kabra2001, 2022-02-22:
Actually it was CentOS issue, I tried installing hls4ml in ubuntu and it worked.

After doing all the steps for acclerator generation and integration correctly. I named my accelerator mlp_3layers_test and while integration in esp it got renamed to mlp_3layers_test_hls4ml. Now in SoC design and testing I used make mlp_3layers_test_hls4ml and I am facing this error. 

[image]

closed 2022-02-23 by mayank-kabra2001