sld-columbia/esp #149: Errors while using command make make mlp_3layers_test_hls4ml-hls in hls4ml accelerator integration

mayank-kabra2001, 2022-02-23:
Actually it was CentOS issue, I tried installing hls4ml in ubuntu and it worked.

After doing all the steps for acclerator generation and integration correctly. I named my accelerator mlp_3layers_test and while integration in esp it got renamed to mlp_3layers_test_hls4ml. Now in SoC design and testing I used make mlp_3layers_test_hls4ml-hls and I am facing this error. 

[image]

[mlp_3layers_test_hls4ml_hls.log](https://github.com/sld-columbia/esp/files/8118413/mlp_3layers_test_hls4ml_hls.log)

_Originally posted by @mayank-kabra2001 in https://github.com/sld-columbia/esp/issues/148#issuecomment-1047986316_

jzuckerman, 2022-02-23:
This is a bug in Vivado HLS that started in 2022. You can fix it with the following patch to `accelerators/vivado_hls/common/hls/common.tcl`:

```
diff --git a/accelerators/vivado_hls/common/hls/common.tcl b/accelerators/vivado_hls/common/hls/common.tcl
index 5300d75..49a1f6a 100644
--- a/accelerators/vivado_hls/common/hls/common.tcl
+++ b/accelerators/vivado_hls/common/hls/common.tcl
@@ -95,7 +95,19 @@ foreach dma $dma_width {
        # cosim_design

        # Export RTL
-       export_design -rtl verilog -format ip_catalog
+       #export_design -rtl verilog -format ip_catalog
+    if { [catch {export_design -format ip_catalog} errmsg] } {
+      puts stderr "export_design exited with error message: $errmsg\n"
+      puts stderr "export_design continuing execution...\n"
+      set filedate "./${ACCELERATOR}_dma${dma}_w${width}/${ACCELERATOR}_acc/impl/ip/run_ippack.tcl"
+      puts $filedate
+      exec /bin/sed -i {s/## Variables/## Variables\nset Revision \[expr \$Revision - 80000000\]/g} $filedate
+      cd ./${ACCELERATOR}_dma${dma}_w${width}/${ACCELERATOR}_acc/impl/ip/
+      puts "Run pack.sh"
+      exec ./pack.sh
+      puts "Done pack.sh"
+      cd ../../../../
+    }
     }
 }
```

mayank-kabra2001, 2022-02-23:
But I have my generated accelerator from hls4ml in accelerators/hls4ml/mlp3layers_hls4ml/hw/hls-work-virtex7/common.tcl. Should I do the changes there ?

GiuseppeDiGuglielmo, 2022-02-23:
This bug affects the Vivado toolchain and HLS in particular, I would suggest applying [this patch from Xilinx](https://support.xilinx.com/s/article/76960?language=en_US). This should fix your issue with hsl4ml and any other design you run with Vivado HLS.

closed 2022-02-25 by mayank-kabra2001