sld-columbia/esp #152: Creating RTL for ASIC target

yyhanafy, 2022-02-25:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

yyhanafy, 2022-02-25:
I browsed through the documentation and tutorials, and could not find any pointer on how to produce a whole RTL of the entire ESP grid of tiles ready for an ASIC process (such as Skywater 130 nm). any pointers on how to do that?

maico-cassel, 2022-02-27:
Hi yyhanafy,
we do not have a stable ASIC flow yet to share with all esp community. However, I can help you with the steps to make your final RTL ready for you proceed with Logic Synthesis and P&R. Before we moving forward, I want to make sure you have the memory compiler for the technology you want to tape-out (Skywater or any other). You are going to need the memory verilog model for simulation and at least the liberty model for Logic Synthesis.
Best,
Maico

yyhanafy, 2022-03-01:
Thank you Maico, this should be a great help.
We are working with OpenRam memory compiler and the Skywater 130nm tech.
From what I understand, the liberty model is included in the Skywater 130nm PDK.

Best Regards
—Y

yyhanafy, 2022-03-01:
Forgot to mention that the implementation tool is OpenLane, an open source RTL to GDSII 
https://github.com/The-OpenROAD-Project/OpenLane

Best Regards
—Y

maico-cassel, 2022-03-03:
Thanks for sharing the OpenLane. First time I heard about it.
I will point you the initial steps you have to do to map your memory models and set the scripts to generate an ASIC RTL.

Step 1:
Inside `esp/socs/epochs0-gf12` you will find a set of files that are technology dependent. You should create a similar folder, lets say `my-sky130`, and copy all contents of `epochs0-gf12` to `my-sky130`.

Step 2:
Modify EPOCHS0-TOP.vhd interface (inputs and outputs) according to the interface of your chip. In addition, modify the PAD instantiation section. The NOC and the Tiles instantiation section you do not need to modify because they are all parameterized according to the parameters file generated by ESP configuration. You can change the name of this file and the Entity according to your design, lets say MY_DESIGN.vhd and MY_DESIGN.

Step 3:
In `my-sky130/Makefile`, change TECHLIB variable for the technology you are using, lets say `sky130` and the CHIP_TOP variable to your top-level design MY_DESIGN. In addition, change every GF12 that exists in this file to your technology name `sky-130`

Step 4:
Update `fpga_proxy_top.vhd`, `top.vhd`, `pads_loc.vhd`, `pads_loc.txt`, `testbench.vhd` according to MY_DESIGN interface.

Step 5:
Inside `esp/rtl/techmap/` there is a folder called `gf12`. You will have to create your tech folder `sky130` and copy the same structure and files of `gf12`. You will have to update all these files according to your technology.

Step 6:
Inside `esp/rtl/techmap/maps`, modify the files `allmems.vhd` and `allpads.vhd` according to your technology

Step 7:
Inside `esp/utils/flist/techmap_vlog.flist`, replace all GF related memories and paths to yours.

Step 8:
This may be the hardest step if you do not have the same memory sizes that already exists in epochs0-gf12. If you run a `grep -r GF12 esp/rtl/caches/*` you will see which cache files instantiate and defines the technology dependent memories. You will have to change these files according to your memories. If the sizes are the same, it is just replace the names. If only the depth are different, you just need to change the definition `GF12_SRAM_SIZE` to match your depth. On the other hand, if your memory sizes are totally different, you will have the change the code to match your sizes with the expected cache size:
Cache size = sets * ways * 16
Where:
L2 Sets: 32 64 128 256 512 1024 2048 4096 8192
L2 Ways: 2 4 8
LLC Sets: 32 64 128 256 512 1024 2048 4096 8192
LLC Ways: 4 8 16

Step 9:
Now, execute `grep -r gf12 esp/rtl/*`. Replace all gf12 related technology by sky130.

Step 10:
Add in ASICLIB variable `sky130` in `esp/utils/make/design.mk`.

Step 11:
In the same level of `esp` directory, create a folder called `sky130`. In this folder create another folder called `verilog`. Copy or create a soft link of your memory verilog models to this folder. In addition, create a folder called `mem`. In this folder you will create the memory wrappers that are instantiated in ESP files (the caches instances) to your technology dependent memory verilog models.

After executing all these changes, your memories should be mapped in the code. The next steps will depend on your clock strategy: internal ring oscillator, internal PLL, or external clock.

Let me know if you get stuck in any part or if something is not clear.

Best,
Maico