sld-columbia/esp #157: Error while doing "make vivado-syn" for single-core Ariane SoC on Xilinx VCU128

ankurlimaye2, 2022-03-28:
I am trying to follow the steps mentioned in [How to: design a single-core SoC](https://www.esp.cs.columbia.edu/docs/singlecore/singlecore-guide/) for the default single-core Ariane SoC with ESP RTL caches on Xilinx VCU128 FPGA board. However, when running the "make vivado-syn", synthesis fails (Error: Failed synthesizing module 'llc_localmem_gf12')

Here's the attached Vivado synthesis log: [vivado_syn.log](https://github.com/sld-columbia/esp/files/8366238/vivado_syn.log)

Desktop (please complete the following information):
 - OS: Ubuntu 18.04.6 LTS
 - CAD tools versions: Xilinx Vivado 2020.2

jzuckerman, 2022-03-29:
ESP officially supports Vivado 2019.2. Can you try again with that version of the tool?

ankurlimaye2, 2022-03-30:
Using 2019.2 resolved the issue. Thank you!

closed 2022-03-30 by ankurlimaye2