sld-columbia/esp #159: Xcelium flow

mayank-kabra2001, 2022-04-20:
Hi, 
I was trying to integrate mlp3layers using the flow mentioned in this(https://www.esp.cs.columbia.edu/docs/hls4ml/).
I would like to confirm whether I am done with make xmsim command. I have attached screenshot of it. 

[image]

Please let me know if I am done with the flow .. If yes, where can I see the report files generated by xcelium , else if this is a issue please let me know. 

Thanks!!

jzuckerman, 2022-04-20:
From here, you can type "run", to start the simulation. It will terminate when the simulation is done (assuming there are no bugs in the design).

mayank-kabra2001, 2022-04-23:
Yes, I ran the run command and it is saying that there is no file to be read... Do I need to put files manually, if yes from where, Will it be generated by the flow? If no, Can you point out the issue? 

[image]

Thanks!!

jzuckerman, 2022-04-25:
What type of file are you looking for? The simulation will show any serial output (printf, etc.) from the software test the SoC is executing. 

If you wish to look at waveforms etc, you can launch the GUI with `make xmsim-gui`.

mayank-kabra2001, 2022-04-25:
I dont know .. In the above screenshot, It is saying not able to find those .dat files. I ran through the normal flow mention in the ESP documentation.

jzuckerman, 2022-04-27:
Seems like those data files are generated as a part of HLS4ML to initialized some memories. Do those files exist somewhere?

mayank-kabra2001, 2022-05-06:
How do we delete some accelerators that we have created using HLS4ML flow and added them in the ESP flow ? I want to delete some accelerators which are redundant in esp-xconfig gui.

closed 2022-05-10 by mayank-kabra2001