sld-columbia/esp #166: "fatal: The remote end hung up unexpectedly" when building RISC-V 32-bit toolchain

LucaUrbinati44, 2022-07-09:
**Describe the bug**
I get a `fatal: The remote end hung up unexpectedly` when launching `./utils/toolchain/build_riscv32imc_toolchain.sh` inside the `esp` folder, as described in the "Software toolchain" paragraph of the ESP documentation: https://www.esp.cs.columbia.edu/docs/setup/setup-guide/.

**To Reproduce**
Steps to reproduce the behavior:
1. Just run `./utils/toolchain/build_riscv32imc_toolchain.sh` inside the `esp` folder.

**Expected behavior**
A clear and concise description of what you expected to happen.
I'm expecting the scripts `build_riscv32imc_toolchain.sh` ends with:
`*** Successfully installed RISC-V (rv32imc) toolchain to $TARGET_DIR ***`

**Screenshots**
Screenshot of the error message:
[image]

**Desktop (please complete the following information):**
 - OS: CentOS7
 - CAD tools versions: not needed for this issue

Any suggestion to solve this issue is appreciated. Thank you

jzuckerman, 2022-07-09:
Hi Luca, please see the work around in my reply on this issue: https://github.com/sld-columbia/esp/issues/164

LucaUrbinati44, 2022-07-12:
Hi Joseph,
I did as you suggested in issue #164 and it worked. 
However, I had to change the URLs that are reported in my previous screenshoot. Apparently, the problem is not related to ESP itself: https://lists.gnu.org/archive/html/qemu-discuss/2017-10/msg00046.html. 
In particular, I had to do the following operations manually (I found the new URLs by searching for the "non-working" URLs on Google):

1) Cloning 
https://gitlab.com/qemu-project/dtc.git 
https://gitlab.freedesktop.org/pixman/pixman.git
into
`${ESP_ROOT}/_riscv32imc_build/riscv-gnu-toolchain/riscv-qemu/`.

2) Cloning
https://gitlab.com/qemu-project/SLOF.git
https://gitlab.com/qemu-project/ipxe.git
https://gitlab.com/qemu-project/openbios.git
https://gitlab.com/qemu-project/openhackware.git
https://github.com/rth7680/qemu-palcode.git
https://gitlab.com/qemu-project/seabios.git
https://gitlab.com/qemu-project/sgabios.git
https://gitlab.com/qemu-project/u-boot.git
https://gitlab.com/qemu-project/vgabios.git
into
`${ESP_ROOT}/_riscv32imc_build/riscv-gnu-toolchain/riscv-qemu/roms/`.

3) Then, I commented [lines 84 and 95](https://github.com/sld-columbia/esp/blob/a8863f80518e6c86ee922445d22c194b827eaf02/utils/toolchain/build_riscv32imc_toolchain.sh#L84) of `${ESP_ROOT}/utils/toolchain/build_riscv_toolchain.sh`, similarly to what is suggested in #164.

4) Rerun the script.

[image]

The strange thing is that I already installed ESP and the "riscv32imc toolchain" on another CentOS7 machine (2 months ago) and I did not encounter this issue.

I hope my contribution can help you to solve this issue.
On my side, the issue can be closed. 
Thanks

jzuckerman, 2022-07-15:
Thanks for the suggestions!

closed 2022-07-15 by jzuckerman