sld-columbia/esp #167: SoC simulation stuck at "Start"

Capoz1898, 2022-07-10:
Hi, I have realized an Accelerator using Catapult HLS but when I try to simulate the complete SoC where I have instantiated my device, the simulation stucks in the while loop where the SoC wait for the DONE signal. I have also checked from the Modelsim waveforms that when I simulate only the accelerator the DONE signal is correctly asserted. I do not know if this is a sort of bug or something else, can someone help me?

biruk-belay, 2022-07-10:
Hi,

Could it be that the accelerator is processing a larger workload and hence
the simulation is slow?

With Regards

- biRuk

Capoz1898, 2022-07-10:
Hi,

thank you for your reply, in order to check this possibility, I tried to put a for loop in the waiting while that after a certain amount of time forces the DONE signal to '1' in order to exit from the while. In this way the accelerator validates correctly the output data with my gold output. This behavior suggest me that the accelerator does his job but in some way the DONE signal is not recognized. Am I wrong?

jzuckerman, 2022-07-10:
Can you visualize the top level of the accelerator in the simulation waveforms? Do you see that the done is ever asserted? That would be the first thing to check. 

You can find the accelerator at this path in the simulation hierarchy: `sim:/testbench/top_1/esp_1/tiles_gen(<tile#>)/accelerator_tile/tile_acc_i/tile_acc_1/<acc_name>_gen/noc_<acc_name>_i/<acc_name>_rlt_i`, where `<tile#>` is replaced by the tile where your accelerator is placed and `<acc_name>` is replaced by the accelerator name.

Capoz1898, 2022-07-10:
Hi @jzuckerman , I tried right now to simulate the entire system and you are right, the done is not asserted in my accelerator when the simulation stucks, but it is asserted when I simulate the accelerator alone. Do you know a possible way to solve this problem? Thank you for your help, I attach also an image from the simulation
[image]

jzuckerman, 2022-07-15:
I would examine what happens on the DMA interface of the accelerator. Are the expected number of words coming into/out of the accelerator? Are all of the expected DMA transactions happening? Are any of the interfaces stuck in a state that is not ready to receive data?

Capoz1898, 2022-07-18:
Thank you @jzuckerman for your reply, I tried to see all the DMA interface signals of the accelerator, I discovered that when I simulate the SoC, what happens is that the in the DMA_read_channel I have only "one half" of the expected signals (I use two PLMs for the input data in my accelerator) and after the first reading it becomes very slow and it does not reach the write process. So after that I do not know why it does not read the second memory signals and maybe this can be the reason why it becomes so slow.

Capoz1898, 2022-07-20:
Hi guys, I want to provide an update, I have partially solved my problem, now the simulation reaches the end, but what happens now, looking at the DMA signals, is that the accelerator does not write in the channel after the reading, and at a certain point it does some strange readings and writings. I try to attach a screenshot, if someone have a solution I will be forever grateful :)
[image]

closed 2022-10-08 by Capoz1898