sld-columbia/esp #197: Cannot find required source file '*_localmem_asic.sv'

SaltedFishLZ, 2023-04-05:
**Describe the bug**
I am trying to use the official Docker image (centos7-full) and the release version 2023.1.0 to do some basic simulations with Xcelium. I found some required files, like 'l2_localmem_asic.sv', are missing. I cannot find them in the source files on GitHub.
'

**To Reproduce**
Steps to reproduce the behavior:
1. 'cd socs/xilinx-vc707-xc7vx485t'
2. 'make esp-xconfig'
3. Change the default CPU to Ibex
4. 'make soft'
5. 'TEST_PROGRAM=./soft-build/ibex/systest.exe make xmsim '

**Expected behavior**
Simulation should run

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: official Docker container on RHEL 8.6
 - CAD tools versions: Xcelium 21.09, Vivado 2019.2

jzuckerman, 2023-04-05:
This looks like incompatible versions between the ESP repo and the esp-caches repo. I think a `git submodule update --init --recursive` will realign them. The simulations should work out of the box in the Docker image. Did you attempt to update the repo in some way that might've caused this?

SaltedFishLZ, 2023-04-05:
I was dealing with another issue that was mentioned in a previous issue:https://github.com/sld-columbia/esp/issues/198#issue-1654795690
I forgot to update recursively. That might be the cause, and let me try it again. Silly me.

SaltedFishLZ, 2023-04-05:
The simulation runs now. Thanks for pointing out my mistake.

closed 2023-04-05 by SaltedFishLZ