sld-columbia/esp #198: Xcelium cannot simulate for single-core(Ibex) SoC

SaltedFishLZ, 2023-04-05:
**Describe the bug**
Xcelium reports an error when trying to run a simulation using the single-core configuration with Ibex.

**To Reproduce**
Steps to reproduce the behavior:
1. 'make esp-xconfig'
2. 'make soft'
3. 'TEST_PROGRAM=./soft-build/ibex/systest.exe make xmsim'

**Expected behavior**
I hope the Xcelium simulation can run.

**Screenshots**
```
xmvhdl_p: *W,ASILDR (/home/espuser/esp/rtl/tiles/tile_io.vhd,688|32): illegal entity designator [5.1].
    xmvhdl  -NOVITALCHECK -linedebug -v93 -nocopyright -work work /home/espuser/esp/rtl/tiles/esp.vhd
    xmvhdl  -NOVITALCHECK -linedebug -v93 -nocopyright -work work /home/espuser/esp/rtl/tiles/asic/asic_tile_acc.vhd
    xmvhdl  -NOVITALCHECK -linedebug -v93 -nocopyright -work work /home/espuser/esp/rtl/tiles/asic/asic_tile_cpu.vhd
    xmvhdl  -NOVITALCHECK -linedebug -v93 -nocopyright -work work /home/espuser/esp/rtl/tiles/asic/asic_tile_empty.vhd
    xmvhdl  -NOVITALCHECK -linedebug -v93 -nocopyright -work work /home/espuser/esp/rtl/tiles/asic/asic_tile_io.vhd
    xmvhdl  -NOVITALCHECK -linedebug -v93 -nocopyright -work work /home/espuser/esp/rtl/tiles/asic/asic_tile_mem.vhd
    xmvhdl  -NOVITALCHECK -linedebug -v93 -nocopyright -work work /home/espuser/esp/rtl/tiles/asic/asic_tile_slm.vhd
    xmvhdl  -NOVITALCHECK -linedebug -v93 -nocopyright -work work /home/espuser/esp/rtl/tiles/asic/asic_tile_slm_ddr.vhd
      haddr           => this_slmddr_haddr,
                                         |
xmvhdl_p: *E,UNMSBD (/home/espuser/esp/rtl/tiles/asic/asic_tile_slm_ddr.vhd,247|41): unmatched bounds in actual and formal subtypes in 93[2.1.1.2],[12.2.4].
      hmask           => this_slmddr_hmask,
                                         |
xmvhdl_p: *E,UNMSBD (/home/espuser/esp/rtl/tiles/asic/asic_tile_slm_ddr.vhd,248|41): unmatched bounds in actual and formal subtypes in 93[2.1.1.2],[12.2.4].
make: *** [xcelium/xmready] Error 1
```

**Desktop (please complete the following information):**
 - OS: Official Docker image (centos7-full) on RHEL 8.6
 - CAD tools versions: Xilinx Vivado 2019.2, Cadence Xcelium 21.09

jzuckerman, 2023-04-05:
This Xcelium issue was fixed in the release a while ago. Are you working on the latest release of ESP? If not, execute a `git pull` followed by a `git submodule update --init --recursive` to get the latest updates. Also, note that we recently (2 weeks ago) updated the ESP docker images. So, if you pull those, they should have the latest versions of ESP already included.

SaltedFishLZ, 2023-04-05:
Solved as mentioned in https://github.com/sld-columbia/esp/issues/197#issue-1654761651

For other users:
Please update and pull the repo recursively.

closed 2023-04-05 by SaltedFishLZ