sld-columbia/esp #203: Problem running simulation in tutorial for designing a single core soc

HelpDesperatelyNeeded, 2023-07-24:
**Describe the bug**
When I try and run "`make run --all`" in ModelSim in the gui, as it tells me to do in the tutorial at
https://esp.cs.columbia.edu/docs/singlecore/singlecore-guide/ 
I get an error saying "unrecognized option '--all'".

**To Reproduce**
I have followed the steps in that tutorial up until that point and run the "`make sim-gui`" command successfully. I then typed "`make run --all`" in the transcript area of the gui.

**Expected behaviour**
I expected it to complete the simulation.

**Continued issue**
When I instead select "run -all" from the simulation drop down menu in the toolbar, after a while of running, I get "Failure: Program completed!" which then says "Break in Process line__225" at the top.vhd file. I have not edited the the top.vhd file at all. How do I solve this problem?

**Desktop**
 - OS: CentOS 7
 - CAD tools versions: Xilinx Vivado 2019.2
 - ModelSim 2019.4

jzuckerman, 2023-07-24:
This is a typo in the guide. You should indeed just execute "run -all" or "run -a". The "Failure: Program completed!" is not a problem. This will always print when the processor has finished running the executable that you have specified. Do you see the expected output from the program you're simulating? The default program should print "Hello from ESP!".

jzuckerman, 2023-07-24:
Yes, the simulation is running properly then.

closed 2023-07-24 by jzuckerman