sld-columbia/esp #205: Trying to make single-core SoC tutorial work for Nexys A7-100T

HelpDesperatelyNeeded, 2023-07-26:
**Describe the bug**
I am following the tutorial at https://esp.cs.columbia.edu/docs/singlecore/singlecore-guide/#fpga-prototyping-with-prebuilt-material and got an error when trying to run behavioural simulation in vivado after changing the constraints files to match the different target board.

The error is as follows:
```
ERROR: [VRFC 10-2063] Module <sram_behav> not found while processing module instance <genblk5[0].genblk1[0].genblk1.mixed_sram> [/home/vboxuser/esp/rtl/caches/esp-caches/llc/rtl/llc_localmem_asic.sv:155]
ERROR: [XSIM 43-3322] Static elaboration of top level Verilog design unit(s) in library work failed.
```
How do I fix this?

**To Reproduce**
I followed the tutorial up until the bit before generating the bitstream. Instead of generating it, I opened the project in vivado and changed the constraint files and the target board to match the Nexys A7-100T. When trying to run a behavioural simulation or synthesise it, I get the above error.

**Expected behavior**
I expect it to run the simulation without issue so that I can generate the bitstream.

**Desktop (please complete the following information):**
 - OS: CentOS 7
 - CAD tools versions: Xilinx Vivado 2019.2
 - ModelSim 2019.4

jzuckerman, 2023-07-26:
It's not finding the sram_behav, which is in a submodule of the Ariane core. Did you clone recursively or run `git submodule update --init --recursive`? If you did, you may need to sync the submodules of Ariane, as we changed them in the last release. You can do this with:
```
cd rtl/cores/ariane/ariane/
git submodule sync
git submodule update --init --recursive
```

However, for FPGA, you should be simulating the FPGA memories and not a behavioral model. This selection is made [here](https://github.com/sld-columbia/esp-caches/blob/d32f1d61495fa4eb20b434147e2b05b6550f97ba/llc/rtl/llc_core.sv#L219-L224). Make sure XILINX_FPGA has been defined in your simulation environment. Notice that we have never used xsim to simulate ESP.

HelpDesperatelyNeeded, 2023-07-28:
I got the simulation to run in ModelSim, but it won't stop running automatically. I think it's related to something I did in trying to port the design to the other board.

I would like help trying to port the design to the Nexys A7-100T board.

I think that I need to create an IP in Vivado using your files first, then instantiate a memory controller or interface and the Ethernet IP in the IP integrator. However, I can't identify the files which should be used to create the first IP.

Do you think any other changes would be required?

Also, do you know what the correct values of L2, LLC and ACC cache set and way numbers for the SoC Generator GUI would be for the Nexys board?