sld-columbia/esp #206: Can't Program Xilinx VC707 Board in Single-Core SoC Tutorial

HelpDesperatelyNeeded, 2023-08-22:
**Describe the bug**
When running the command `FPGA_HOST=localhost XIL_HW_SERVER_PORT=3121 make fpga-program`, I get the following error:
```
    INFO generating programming script for XC7VX485T
INFO: [Labtools 27-2285] Connecting to hw_server url TCP:localhost:3121
INFO: [Labtools 27-2222] Launching hw_server...
INFO: [Labtools 27-2221] Launch Output:

****** Xilinx hw_server v2019.2
  **** Build date : Nov  6 2019 at 22:13:42
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

INFO: [Labtools 27-3415] Connecting to cs_server url TCP:localhost:3042
INFO: [Labtools 27-3417] Launching cs_server...
INFO: [Labtools 27-2221] Launch Output:

****** Xilinx cs_server v2019.2.0
  **** Build date : Nov 07 2019-05:41:48
    ** Copyright 2017-2019 Xilinx, Inc. All Rights Reserved.

Connected to localhost
Searching for XC7VX485T...
ERROR: [Labtoolstcl 44-199] No matching targets found on connected servers: localhost
Resolution: If needed connect the desired target to a server and use command refresh_hw_server. Then rerun the get_hw_targets command.
ERROR: [Common 17-39] 'get_hw_targets' failed due to earlier errors.

    while executing
"get_hw_targets "
    invoked from within
"foreach cable [get_hw_targets ] {
    open_hw_target $cable
    set dev [get_hw_devices]
    if [string match -nocase "$part*" $dev] {
	puts "Programm..."
    (file "program.tcl" line 11)
    INFO Waiting for DDR calibration...
```
Do you have any idea what is causing this, or how to fix it?

**To Reproduce**
I followed the tutorial at https://esp.cs.columbia.edu/docs/singlecore/singlecore-guide/ and used the board mentioned. When I got to the stage about FPGA programming, I connected the UART and JTAG cables from the board to my computer (also tried it with Ethernet connection too) and then ran the above command.

**Expected behavior**
I expected it to successfully program the board.

**Desktop (please complete the following information):**
 - OS: CentOS 7.6 in a virtual machine (I passed through the USB devices to the VM)
 - CAD tools versions: Vivado 2019.2, ModelSim 2019.4

HelpDesperatelyNeeded, 2023-08-23:
When I try and auto-detect a target in Vivado in windows (the host machine), the board shows up. However, when I try to do the same in the VM, it does not show up.

I have done this with a different board in a different VM in the past and it worked properly. Do you think there may some driver I need or is there anything else you can think of that could help?

HelpDesperatelyNeeded, 2023-08-24:
I can successfully program the board from outside the VM, in Windows, by copying the Vivado Project to the host machine and opening it from Vivado 2019.2 there.

However, once I set up Minicom (back inside the VM - it does detect the USB cables), I went to the next section called "Testing on FPGA" and tried running the commands mentioned. From either commands (both `make fpga-run` and `make fpga-run-linux`), I get the exact same response:
```
    CC esplink
    OBJCP /home/vboxuser/esp/socs/xilinx-vc707-xc7vx485t/soft-build/leon3/ram.srec
    OBJCP /home/vboxuser/esp/socs/xilinx-vc707-xc7vx485t/soft-build/leon3/systest.bin
ESPLink address 192.168.1.7:46392
```
It then seems to just hang there. Nothing is output in the Minicom. Do you have any idea what could be wrong here?

I have tried this both with and without Ethernet connected between the board and PC, as I am not sure which is needed. Could you please tell me which cables (UART, JTAG or Ethernet) are needed for this stage?

jzuckerman, 2024-03-14:
JTAG (FPGA programming), UART (console output), and Ethernet (ESPLink) cables are all needed for ESP. For ESPLink, make sure that the Ethernet interface you are using on your PC is in the same subnet as the IP address shown. by ESPLink (i.e. 192.168.1.X).

closed 2024-03-14 by jzuckerman