sld-columbia/esp #208: Missing NV_NVDLA.verilog and Collecting generated RTL files

yccai36, 2023-10-11:
Hello there,

I want to use my FPGA-based EDA tool (similar with  Zebu from @Synopsys) to run a single-core SoC. It utilizes Xilinx Virtex Ultrascale _XCVU440_ FPGA boards.

My current step is to follow the tutorial on the website: [https://www.esp.cs.columbia.edu/docs/singlecore/singlecore-guide/#fpga-prototyping](url) and ran the `make esp-xconfig` command under the `esp/socs/xilinx-vcu118-xcvu9p` folder.

There are two questions I want to ask:

1. **What should I do here to extract the RTL files for my generated SoC design**? A filelist of the SoC design is mandatory for my EDA flow.

2. At the start of simulation, the terminal reports a message below:

>   strings: '[my machine path]/esp/accelerators/third-party/NV_NVDLA/NV_NVDLA.verilog': No such file

Is anything going wrong with my setup environment?

Thanks for your help!

Best,
Yuchen Cai

jzuckerman, 2023-10-11:
1. You can run the command `make genus-setup` from your SoC design folder. This will create two scripts inside a folder called `genus`: `srcs.tcl` and `incdir.tcl`. These contain all of the RTL sources and include directories, respectively. 

2. Did you run `make NV_NVDLA` before running the simulation?

yccai36, 2023-10-12:
Thanks for your swift reply Jzuckerman!
> You can run the command make genus-setup from your SoC design folder. This will create two scripts inside a folder called genus: srcs.tcl and incdir.tcl. These contain all of the RTL sources and include directories, respectively.

At which step during ESP flow, should I ran the `make genus-setup` to get the complete RTL filelist?

 > Did you run make NV_NVDLA before running the simulation?

I  wish to try **a single-core SoC design with the NVDLA accelerator** and below is my expected ESP flow: 
1. `cd esp/socs/xxx/`
2.  `make esp-xconfig` and this step will produces the configuration file.
[image] 
* Is the "caches" option mandatory when using NVDLA? If I choose to use "ESP RTL" implementation for the caches, are all of the caches files produced in .sv format?
3. `make NV_NVDLA`:
* Is this make target needed to run evertime?  
5. `make sim` and `run -all`:
* Can these two simulation steps be skipped since I only have VCS simulator at hand?
6. `make vivado-syn`
7. `make soft` and `make linux` :
* The first command will produce **prom.bin** and **systest.bin** while the latter one produce the Linux image **linux.bin**. Can I put them into one SD card and boot the system automatically when clock and reset is on? Besides, it is said that the SoC runs at 50MHz but my FPGA board can only run 10MHz in maximum. Will the reduction of clock frequency cause problems?

closed 2024-03-14 by jzuckerman