sld-columbia/esp #233: DTC and vivado_hls errors with bump to Vivado2023.2

kcanidaISI, 2024-08-10:
Bumping to Xilinx Vivado2023.2 in our current container environment is failing with a couple of errors.

**Desktop:**
 - OS: ESP centos7-full docker image
 - CAD tools versions: Xilinx Vivado 2023.2
 
### DTC Error:
**To Reproduce**

        1. Within docker container
        2. 'source esp_env_cad.sh'
        3. 'cd esp/socs/xilinx-vc707-xc7vx485t'
        4. 'make esp-defconfig'
        5. 'make soft'

[image]

**Additional context**
Sourcing Vivado 2023.2 settings64.sh adds newer dtc version installed with Vitis to path

[image]

### HLS Error:
**To Reproduce**

        1. 'make adder_vivado-hls'
 
[image]

**Additional context**
vivado_hls deprecated in Vivado 2023.2 for vitis_hls

jzuckerman, 2024-09-10:
With Vivado 2023.2, vivado_hls, has changed to vitis_hls. You could try patching [this line](https://github.com/sld-columbia/esp/blob/new-plic/accelerators/vivado_hls/common/hls/Makefile#L11) with that change. I am not sure, though, if there will be other issues.

Alternatively, you could use a different version of Vivado to run HLS (I'd suggest 2019.2, which is what we've tested with) than for generating bitstreams.

jzuckerman, 2024-09-10:
For the DTC issue, I would suggest just using the one installed on your machine than the one from Vivado. When I run `which dtc` I get back `/usr/bin/dtc`.